finitespace/BME280 #1: Help wanted...

idesignstuff, 2016-01-24:
Hi Tyler.

Thanks for putting this together. Works like a champ. I'm in my first weekend messing with the world of Arduino. So far, I've gotten: 2 nRF24L01+ to talk,  the DHT-11 to return data, the BME280 to return data(thanks to your library and example sketch), and all on Pro Mini knockoffs. My question is this.

How (or where) do I change the units for the pressure here?
/* ==== Functions ==== */
void printBME280Data(Stream* client){
  float temp(NAN), hum(NAN), pres(NAN);
   uint8_t pressureUnit(3);   // unit: B000 = Pa, B001 = hPa, B010 = Hg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
  bme.ReadData(pres, temp, hum, metric, pressureUnit);                // Parameters: (float& pressure, float& temp, float& humidity, bool hPa = true, bool celsius = false)
 

Specifically, I'm lost as to where to put "B010" in the code.
Also, not sure what the  "   uint8_t pressureUnit(3);  " is doing.

I've spent most of the weekend reading arduino.cc learning resources, and scouring google for explanations.

FWIW, I'm experienced with wiring, basic electronics, and hardware, but a total noob with c++.
My only coding experience has been Apple BASIC in Jr. High, and dabbling in VB for M$ Office.

Any advice? 
Thanks again for writing this library!

idesignstuff, 2016-01-24:
Nevermind, I got it figured out.

/* ==== Functions ==== */
void printBME280Data(Stream* client){
  float temp(NAN), hum(NAN), pres(NAN);
   uint8_t pressureUnit(B010);   // unit: B000 = Pa, B001 = hPa, B010 = Hg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
  bme.ReadData(pres, temp, hum, metric, pressureUnit);                // Parameters: (float& pressure, float& temp, float& humidity, bool hPa = true, bool celsius = false)

Still not sure what to do with oversampling though...

finitespace, 2016-01-28:
Hello! Thanks, I am glad to hear it is getting used!

There might be some slight confusion because I forgot to update the comment in the example code (just changed). As you probably found out the comment shows binary values correlating to the pressure units. The desired pressure unit can be specified by passing a numerical value in through the parameter pressureUnit by converting the Binary to Hex or Decimal.

Oversampling is well described on the wiki: https://en.wikipedia.org/wiki/Oversampling.
More specific details about the oversampling performed on the BME280 can be found in the BME280 data sheet: http://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BMP280-DS001-12.pdf
on page 15 under each measurement, and page 18 under Noise.

On the wiki here is a good explanation:
"In practice, oversampling is implemented in order to achieve cheaper higher-resolution A/D and D/A conversion.[1] For instance, to implement a 24-bit converter, it is sufficient to use a 20-bit converter that can run at 256 times the target sampling rate. Combining 256 consecutive 20-bit samples can increase the signal-to-noise ratio at the voltage level by a factor of 16 (the square root of the number of samples averaged), effectively adding 4 bits to the resolution and producing a single sample with 24-bit resolution.[3]"

So to sum it up, oversampling is the term used for collecting a number of samples and then getting the average. The advantage is greater accuracy and less noise. The disadvantage is increased power consumption. 

Does that help?

-Tyler

closed 2016-02-05 by finitespace

idesignstuff, 2016-02-05:
Tyler, thanks again for the explanations. Now, can you point me to how to load variables (float? Long? Int?) with values from the bme library, to ready them for sending via rf24's?

idesignstuff, 2016-02-05:
I figured out how to load individual floats with temp, hum, and pres, but I think I need an array or struct.

finitespace, 2016-02-06:
I am not sure I understand your question. Upload your code and I will take a look at it.