finitespace/BME280 #2: compile errors

aaronj314, 2016-02-05:
Sorry for the newbie question.  I've installed the lib per the README and I'm using ID 1.6.5 but I'm getting compilation error using the test example:

Arduino: 1.6.5 (Mac OS X), Board: "Arduino Uno"

Build options changed, rebuilding all
Using library BME280 in folder: /Applications/Arduino.app/Contents/Java/libraries/BME280 (legacy)

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard -I/Applications/Arduino.app/Contents/Java/libraries/BME280 /var/folders/cf/6c8fqtkj4b91nhzcrfhlzz5h0000gn/T/build8708549420658519769.tmp/BME280_Test.cpp -o /var/folders/cf/6c8fqtkj4b91nhzcrfhlzz5h0000gn/T/build8708549420658519769.tmp/BME280_Test.cpp.o 
In file included from BME280_Test.ino:37:0:
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:48:3: error: 'uint8_t' does not name a type
   uint8_t tempOversamplingRate, humidityOversamplingRate, pressureOversamplingRate; // B000 = Skipped, B001 = x1, B010 = x2, B011 = x4, B100 = x8, B101/other = x16
   ^
In file included from BME280_Test.ino:37:0:
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:49:3: error: 'uint8_t' does not name a type
   uint8_t mode;                               // Sleep = B00, Normal = B01 and B10, Forced = B11
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:50:3: error: 'uint8_t' does not name a type
   uint8_t standbyTime;                        // B000 = 0.5ms, B001 = 62.5ms, B010 = 125ms, B011 = 250ms, B100 = 250ms, B101 = 1000ms, B110 = 10ms, B111 = 20ms
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:51:3: error: 'uint8_t' does not name a type
   uint8_t filter;                             // B000 = off, B001 = 2, B010 = 4, B011 = 8, B100/other = 16
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:52:3: error: 'uint8_t' does not name a type
   uint8_t controlHumidity;                    // ctrl_hum register. (ctrl_hum[2:0] = Humidity oversampling rate.)
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:53:3: error: 'uint8_t' does not name a type
   uint8_t controlMeasure;                     // ctrl_meas register. (ctrl_meas[7:5] = temperature oversampling rate, ctrl_meas[4:2] = pressure oversampling rate, ctrl_meas[1:0] = mode.)
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:54:3: error: 'uint8_t' does not name a type
   uint8_t config;                             // config register. (config[7:5] = standby time, config[4:2] = filter, ctrl_meas[0] = spi enable.)
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:55:3: error: 'uint8_t' does not name a type
   uint8_t bme_280_addr;
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:56:3: error: 'uint8_t' does not name a type
   uint8_t dig[32];
   ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:60:22: error: 'uint8_t' has not been declared
   void WriteRegister(uint8_t addr, uint8_t data);
                      ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:60:36: error: 'uint8_t' has not been declared
   void WriteRegister(uint8_t addr, uint8_t data);
                                    ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:66:17: error: 'int32_t' has not been declared
   bool ReadData(int32_t data[8]);
                 ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:69:30: error: 'int32_t' has not been declared
   float CalculateTemperature(int32_t raw, int32_t& t_fine, bool celsius = true);
                              ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:69:43: error: 'int32_t' has not been declared
   float CalculateTemperature(int32_t raw, int32_t& t_fine, bool celsius = true);
                                           ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:72:27: error: 'int32_t' has not been declared
   float CalculateHumidity(int32_t raw, int32_t t_fine);
                           ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:72:40: error: 'int32_t' has not been declared
   float CalculateHumidity(int32_t raw, int32_t t_fine);
                                        ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:75:27: error: 'int32_t' has not been declared
   float CalculatePressure(int32_t raw, int32_t t_fine, uint8_t unit = 0x0);   // unit: B000/other = Pa, B001 = hPa, B010 = inHg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
                           ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:75:40: error: 'int32_t' has not been declared
   float CalculatePressure(int32_t raw, int32_t t_fine, uint8_t unit = 0x0);   // unit: B000/other = Pa, B001 = hPa, B010 = inHg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
                                        ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:75:56: error: 'uint8_t' has not been declared
   float CalculatePressure(int32_t raw, int32_t t_fine, uint8_t unit = 0x0);   // unit: B000/other = Pa, B001 = hPa, B010 = inHg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
                                                        ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:79:18: error: expected ')' before 'tosr'
   BME280(uint8_t tosr = 0x1, uint8_t hosr = 0x1, uint8_t posr = 0x1, uint8_t mode = 0x3,
                  ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:90:22: error: 'uint8_t' has not been declared
   float ReadPressure(uint8_t unit = 0x0);   // unit: B000 = Pa, B001 = hPa, B010 = Hg, B011 = atm, B100 = bar, B101 = torr, B110 = N/m^2, B111 = psi
                      ^
/Applications/Arduino.app/Contents/Java/libraries/BME280/BME280.h:96:85: error: 'uint8_t' has not been declared
   void  ReadData(float& pressure, float& temp, float& humidity, bool metric = true, uint8_t p_unit = 0x0);
                                                                                     ^
Error compiling.

finitespace, 2016-02-05:
Hey there!

I pushed a change to trunk to run legacy versions of Arduino. Pull down the latest version and give it a go!

Tyler

aaronj314, 2016-02-05:
That worked!  Thx!.

closed 2016-02-05 by aaronj314