finitespace/BME280 #3: How to change the standby time?

canhub, 2016-04-30:
How to change the standby time?
I need to take every 250ms one sample.

finitespace, 2016-04-30:
Standby mode can be changed in the BME280 constructor. 

However, standby time is only used in normal mode; in forced mode, which is the default, the measurements are taken at whatever rate you ask for them.

canhub, 2016-05-01:
Hi! Thanks for the reply!
everything works ok.
Is there an easier way to change the standby time and/or the mode, within my code not from the library.

finitespace, 2016-05-02:
You're welcome.

Yes, when you call the constructor in your code, the standby time is the 5th argument.

For example:

`BME280 bme(1, 1, 1, 1, 4); // Would set the standby time to 250ms and put the sensor in normal mode.`

closed 2016-05-04 by canhub