finitespace/BME280 #4: Temperature question

TheAustrian, 2016-05-10:
Hi! First of all thanks for this library, it's a great job you did here!

I just wanted to ask, since in my experience the BME reads temperature a bit higher than some other sensors, if you compensate for that in the library? 

I'm asking because at the moment I'm using a -2°C offset in the php file that fetches the data from my database but I noticed that when I changed to your library from the Sparkfun one I got a drop in the temperature readings - so it seems your library is closer to the "real" values and I might need to adjust my offset.

Thanks, Alex

finitespace, 2016-05-11:
Hey thanks!

I have noticed that the BME reads a little differently than my DS18B20 temp probes. To answer your question right off the bat, no, the library does not use any sort of hard coded offset to modify the temperature.

With that said, each probe is calibrated at the factory and these 'trim' values are stored in non-volital memory on the device. The temperature values are returned from the BME probe as a 32-bit raw data, which is then bitwise-or with the trim values to calibrate the temperature. 

I believe there are two algorithms provided by Bosch, one to perform fixed point calibration and one to determine double floating point calibration.

The fix point is the harder and longer of the two algorithms and provides an exact conversion from the raw data to celsius and is the one used in this library.

The double floating point does a series of (double) casts a long the way and gives an approximation of the temperature.

All that being said, I have not looked at the Sparkfun library, but the reason you could be seeing different results is for any of the following (or something else entirely): First, they could be ignoring the trim entirely. Second, they could be doing the calibration wrong or casting away precision. Third, they could be using the less precise way of calculating the temperature.

Hope this helps!
Tyler

TheAustrian, 2016-05-11:
Thanks a lot! Yes, it helps & is quite interesting! If calibration is done at the Bosch factory I can imagine the PCB of the breakout boards also might play some role (if the sensor has some self-heating), since that assembly happens afterwards.

Thanks again for the library ;)

closed 2016-05-11 by TheAustrian

jestani, 2016-09-02:
MySQL database does not fill up the data from the sensor. I checked everything. Home chart.html has the same name without charts. :(.... any suggestions?
Jack

finitespace, 2016-09-02:
moving this to it's own issue @jestani #6