finitespace/BME280 #5: ESP8266 crash

ido1990, 2016-06-03:
Hi!
Thanks for this awesome library!

I have a problem,
When I upload the example with Wire.pins (0,2) and then Wire.begin, It work great.
But when I upload my own code (send the data from BME280 to thingspeak) it crash immediately.
When I comment out bme.begin() it work and connect to the WiFi.
So it seems that bme.begin() cause the problem here.
Any ideas?

Thanks!

finitespace, 2016-09-02:
Hi,

So what I am hearing is when you log the readings to a serial interface the BME280 library performs as expected, but when you upload those readings to thingspeak the arduino crashes?

Do you have an online repository of your code I can look at?

Thanks,
Tyler

ido1990, 2016-09-04:
Hi Tyler,
No, the ESP8266 crashes in the moment it calls to bme.begin()
I'm working with the ESP8266 directly with the arduino IDE.
No Arduino actually involved.
I'm using:
https://github.com/esp8266/Arduino

Thanks!

finitespace, 2016-09-12:
Hi ido,

I haven't had time to look at your specific case. I have pulled in a request to merge that includes a solution for esp. See if this resolves your issue. 

Tyler

ido1990, 2016-09-13:
Ok,
Thank you Tyler!

closed 2016-09-19 by finitespace