finitespace/BME280 #8: Connecting multiple BME280s

redsherpa, 2016-09-17:
Well, problem is as simple as that... I'm trying to gather data off of two of these. The code works for one, but I have no idea how to connect a second one up. :) Thanks!!!

finitespace, 2016-09-20:
Hello Red,

To answer your question, yes you can hook up multiple bme280's. 

There are several configurations:

- Two BME280s on one I2C (involves changing the I2C address of one of the sensors)

- Multiple BME280s using a SPI interface (currently working on supporting SPI in version two of the lib (: whahooo you inspired a new revision))

- Finially, all sorts of funk with both I2C and SPI interfaces driving multiple and many thriving BME280s.

I am currently updating the lib for version two (SPI support). Let me know if you would guinea pig for the new SPI interface.

Cheers mate!
Tyler

redsherpa, 2016-09-28:
Thank you, Tyler. Sorry sorry sorry for the lateness of my reply!

I ended up researching to the no-end and going with the first route (soldering the jumper on the body of the 280) which moved it to a different I2C address. Luckily. :)

Sounds awesome that this question resulted in an update! However, I don't know if I can guinea anymore, seeing as how the test I was modifying these for is running and I don't have 2 same-I2C 280s anymore! :)

Thanks again for all the help - you're making a lot of people's lives better!

-Gene

finitespace, 2016-12-19:
Happy Holidays!

Check out the new SPI support.

closed 2016-12-19 by finitespace