finitespace/BME280 #10: Platformio library version bump

chacal, 2016-11-20:
Hi!

Would it be possible to bump up the version string in `library.properties` to 2.0.0 to match the latest release?

It seems that the version Platformio currently distributes is somehow broken as I get the following compile error:

> BME280I2C.h:56:23: error: expected ')' before 'tosr'

By looking at the commits in this repository, it seems that the bug has already been fixed, but the updated version is not available from platformio. This might get resolved if the library version would be updated.

chacal, 2016-11-20:
Just FYI.. By cloning the latest version of the repository the compile error disappears. So we'd just need platformio to provide the latest version as well.

finitespace, 2016-11-21:
Done. Give it a roll now.

chacal, 2016-11-22:
Just tested it and works perfectly. Thanks! :)

closed 2016-11-22 by chacal