finitespace/BME280 #12: SPI Support still missing?

Eheran1, 2016-12-10:
I tested a few librarys, they all lack features of the chip. This is the best one but it lacks SPI support...
Sparkfuns has a low resolution output for some reason while adafruits library cant be set to use any oversampling etc.

U mentioned adding that in the other issue, that would be really great. Otherwise I have to stick to adafruits very basic library...

Also, to have a better overview of the options:

BME280 bme(1, 2, 5, 3, 7, 4); 
// 1 = oversample pressure ×1
// 2 = oversample temperature ×2
// 5 = oversample humidity ×16
// 3 = forced readout (1+2=normal, 0=sleep)
// 7 = 20ms sleeptime
// 4 = filter x16

And a list of the options:
  // 0 = no oversampling
  // 1 = 1x
  // 2 = 2x
  // 3 = 4x
  // 4 = 8x
  // 5 = 16x

  // 0 = Sleep mode
  // 1 or 2 = Forced mode
  // 3 = Normal mode

  // 0 = 0,5ms
  // 6 = 10ms
  // 7 = 20ms
  // 1 = 62,5ms
  // 2 = 125ms
  // 3 = 250ms
  // 4 = 500ms
  // 5 = 1000ms

  // 0 = Filter off
  // 1 = 2x
  // 2 = 4x
  // 3 = 8x
  // 4 = 16x

finitespace, 2016-12-12:
Hi Eheran1,

Thank you for reminding me. I started the reformation for SPI, but never finished because none of my BME boards have SPI headers. I just ordered a board off of amazon and I have a new project to use it with. So I will try to get SPI support up shortly. I will also check to see if anyone has forked the library to add on SPI support. I am thinking it shouldn't be monumental, so I think it might be doable shortly.

 I will keep you updated,

Tyler

Eheran1, 2016-12-12:
Very nice, thank you. 

Thing is I wanted I2C but got SPI...
But even with the basic adafruite lib its crazy how accutare and/or high resolution the readings are.  Well done TI.

finitespace, 2016-12-19:
Happy Holidays!

Check out the new SPI support.

Tyler

closed 2016-12-19 by finitespace

Eheran1, 2016-12-19:
Awesome! Thank you.