finitespace/BME280 #13: Compilation fails with current HEAD

maru-sama, 2016-12-21:
Hello,

First of all, thank you very much for this nice library, it makes working with this sensor a breeze.
I am trying to use current HEAD with platformio and I am facing compilation issues. First of all you are including header file which does not exist in both [BME280Spi.cpp](https://github.com/finitespace/BME280/blob/master/src/BME280Spi.cpp#L36) and
[BME280SPiSw.cpp](https://github.com/finitespace/BME280/blob/master/src/BME280SpiSw.cpp#L35). 
I think this is an accidental left over. After removing this I get the following error.

```
Compiling .pioenvs\huzzah\lib\BME280\BME280SpiSw.o
.piolibdeps\BME280\src\BME280SpiSw.cpp:155:101: error: default argument given for parameter 5 of 'BME280SpiSw::BME280SpiSw(uint8_t, uint8_t, uint8_t, uint8_t, 
uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)' [-fpermissive]

```
This can be fixed by removing the defaults from the .cpp file, since they are already in the .h file. 
After this cleanup it compiles without any issues. Initially I wanted to send a pull request, but I am currently not in front of my develop machine.

finitespace, 2016-12-21:
Hello Maru-sama,

Thank you for the pull.

Great catch. Mine shouldn't have compiled with those defaults in the source. What compiler are you using?

Tyler

maru-sama, 2016-12-21:
I am using platformio for development. Apparently the compiler which is used for esp8266 boards is apparently the [xtensa toolchain](https://github.com/jcmvbkbc/gcc-xtensa)

closed 2016-12-21 by maru-sama

idesignstuff, 2016-12-21:
So I just downloaded the 2.1 version using platform io and compiled an empty sketch with only includes and empty setup/loop functions. I got an error calling the adafruit BME lib. I've never installed that lib. What is happening?

finitespace, 2016-12-21:
idesignstuff,

I had a hell of a time debugging SPI and at one point tried ada's library to check if I was init the bme properly. I forgot to pull the include out before committing. This has been fixed. Try 2.1.01.

Tyler

idesignstuff, 2016-12-21:
You, sir, are the man.

Thanks very much!

Christian

idesignstuff, 2016-12-21:
FYI, I just compiled successfully with the latest. Adafruit is no longer called.
Rock on.

idesignstuff, 2017-01-03:
Tyler,

Please forgive my ignorance, but I'm working with Wemos D1 Mini dev boards,
and they are based on ESP8266 chips.

I'm trying to build a web-connected weather station with a relay to turn
on/off my humidifier based on ambient temp.

How do I configure the BME280 over I2C for an ESP board? I see in
BME280I2C.h:

  /* ==== Method used at start up to initialize the class. Starts the I2C
interface. ==== */
  virtual bool begin();
#if defined(ARDUINO_ARCH_ESP8266)
  /* ==== On esp8266 it is possible to define I2C pins ==== */
  bool begin(int SDA, int SCL);
#endif

But.. I'm not sure where to put:

#define ARDUINO_ARCH_ESP8266

--or where to define the SDA and SCL pins.

Am I missing something?

Christian

theopensourcerer, 2017-01-07:
Hi,

I am having a similar issue to @idesignstuff in the above comment. I am trying to test the BME_280_I2C_Test example. I am using PlatformIO on Ubuntu 16.10 and I have a regular ESP8266 12F connected via a FTDI cable to USB.

I had to include the Arduino SPI library files and add an #include <SPI.h> in my "main.cpp" file just to get it to compile (the include in BME280Spi.cpp was not finding SPI.h). 

I can upload the built code to my ESP but every 500ms it reports it is unable to find the BME Sensor. I tried the default I2C pins GPIO4 & GPIO5. I have also tried adding some extra defines as below:

```
#include <BME280I2C.h>
#include <Wire.h> // Needed for legacy versions of Arduino.
#include <SPI.h>
 
#define ARDUINO_ARCH_ESP8266 1
#define SERIAL_BAUD 115200
#define SDA 4
#define SCL 5
```
But it doesn't seem to help. I guess I'm doing something pretty stupid but C isn't my strong point ;-)

FYI - I have confirmed the BME280 is working using my Raspberry Pi and some Python code:

```
pi@brutha:~ $ python bme-test.py 
Chip ID     : 96
Version     : 0
Temperature :  22.97 C
Pressure :  1019.8970526 hPa
Humidity :  79.9214676864 %
```
TIA

theopensourcerer, 2017-01-07:
Argghhh! Sorry for the noise. After trying 3 other BME/I2C libraries - with no success - I switched to different ESP chip. That one works :-( I must have a dodgy GPIO 4 or 5 on that other chip! Your example is running fine - thanks and apologies again.

idesignstuff, 2017-01-07:
I got around the problem by adding #include <SPI.h>to my .ino file.
Still seems odd to require SPI for i2c.

Christian

finitespace, 2017-01-09:
That does seem odd. I will have to investigate further. I have not yet used the BME with an esp yet, so I will get to that soon.

I am glad you were able to resolve the issues. 

Tyler

idesignstuff, 2017-01-09:
Wemos D1 Mini or NodeMCU would be good, fairly popular boards to test.
Wemos D1 Minis are about $3.

Christian