finitespace/BME280 #15: Change SDA/SCL

theopensourcerer, 2017-01-09:
Hi,

I note at the end of the file BME280I2C.h:

#if defined(ARDUINO_ARCH_ESP8266)
  /* ==== On esp8266 it is possible to define I2C pins ==== */
  bool begin(int SDA, int SCL);
#endif

I wanted like to use this facility to change the SDA/SCL pins used on my ESP8266. I have tried adding #defines at the top of my main.cpp (Am using PlatformIO) which compiled but didn't seem to make any difference in that the BME was still being read on GPIO 4/5.

I did work out how to do this in the end... In the .platformio/packages/framework-arduinoespressif8266/variants/ directory are header files for all the different board types. Once I had worked out which board I was using - it was called nodemcu - (I had actually chosen "Espressif ESP8266 ESP-12E" from the Project builder in PlatformIO) I could set the pins in the appropriate pins_arduino.h file.

I now have my BME280 being read on GPIO 12 & 14 :-) 

Maybe this info will help someone else.

Thanks for a cool library.

idesignstuff, 2017-01-09:
That's very helpful. Thanks Alan!

Christian

idesignstuff, 2017-01-16:
I would like to add wind chill and heat index functions to go with dew point and altitude.

finitespace, 2017-01-17:
Christian, 

Certainly, do you want to code that and submit a pull request? Or are you asking for support? I have ordered a Wemos board and plan to test the BME against it as soon as I have cycles to spare. I could probably add in wind chill and head index at that time.

Tyler

idesignstuff, 2017-01-17:
Tyler, I'll give the code a shot, but I'm pretty close to the beginner end
of the spectrum.

Christian

idesignstuff, 2017-01-18:
> IMHO windchill and heat index should not be in this library. You could
> fork it into an extended library, so we could keep the basic library small
> and simple. The library should only provide the api, which is based on the
> native sensor interface.

Agreed that a simple light lib would be ideal, but the computed functions
seem incomplete with only altitude and dewpoint
Maybe there should be a stripped lib for only basic register reading and
power commands, and then a lib for weather data calculations. Maybe there
already is one of those...

Christian

finitespace, 2017-01-23:
Christian,

It might be worthwhile to make a library just for weather calculations. I have not seen one.

Tyler

idesignstuff, 2017-01-23:
I can give it a shot, but it's going to take me a while to learn enough to
build it. Never written a lib before.

Christian

finitespace, 2017-03-06:
Christian,

Any progress? I was thinking about creating a new library for weather calculations and moving the dew point calculation to that. Just haven't gotten around to it. Sounds like you want heat index and wind chill. Any other calculations I should add. I am trying to come up with 5 or 6 before I put it into action.

Tyler

idesignstuff, 2017-03-06:
Tyler, no, not yet. Dew point, heat index, wind chill sound like they'd be
popular.

Christian

coelner, 2017-08-06:
back to topic. Do not call Wire.begin() at all. It is not the job of a library to init a bus system, only init the specific sub system.
See here: https://www.arduino.cc/en/Reference/WireBegin
and here: https://github.com/esp8266/Arduino/issues/2774

finitespace, 2017-08-07:
Coelner,

I agree, we will have to throw some code in it's place to check initialization instead. I will try to make that change when I make the next interface changes.

Tyler

coelner, 2017-10-07:
First of all, try to use the begin(int SDA, int SCL) call. I guess that the esp32 needs also a assignment for those pins.

Fbo06, 2017-10-07:
Hello cleaner thank you for your response, in fact everything was OK until i upgrade to the 2.1.4 BME280 library. if i correctly understand what you say this begin(int SDA, int SCL) is done in the "Arduino/hardware/espressif/esp32/variants/esp32/pins_arduino.h" file

Fbo06, 2017-10-07:
in fact this issue occurs only with the ESP32 board, with the ESP8266 it compile without errors

coelner, 2017-10-07:
@Fbo06 yeah, look in the other issue #34 
the closing #endif is at the wrong position.

finitespace, 2017-10-08:
All of this should be fixed in version 2.2.

Christian, you can now add any extra calculation functions to the EnvironmentCalculations files.

Tyler

closed 2017-10-08 by finitespace