finitespace/BME280 #16: Error on default startup

Bwanna, 2017-01-22:
Hi,

I'm an avg user, so this could be something I'm doing.  But here's what's happening.

1. Downloaded from gitHub.
2. Unzipped into /Libraries  location
3. Opened "BME280"BME.ino"
4. Verify/Compiled and received the following errors.

If you have a chance, sure could use to know if this is something I did/didn't do, something wrong with my setup, etc.  Other sketches work fine i.e Sparkfun, Adafruit, etc.  Wanted to try yours considering the functions you created.

Thanks,
Steve

____________________________________________________________________
Arduino: 1.8.0 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

C:\Users\Steve\Documents\Arduino\libraries\BME280\examples\BME280_Modes\BME280_Modes.ino: In function 'void printBME280Data(Stream*)':

BME280_Modes:96: error: no matching function for call to 'BME280I2C::ReadData(float&, float&, float&, uint8_t&, bool&)'

   bme.ReadData(pres, temp, hum, pressureUnit, metric);                // Parameters: (float& pressure, float& temp, float& humidity, bool hPa = true, bool celsius = false)

                                                     ^

C:\Users\Steve\Documents\Arduino\libraries\BME280\examples\BME280_Modes\BME280_Modes.ino:96:53: note: candidate is:

In file included from C:\Users\Steve\Documents\Arduino\libraries\BME280\examples\BME280_Modes\BME280_Modes.ino:26:0:

C:\Users\Steve\Documents\Arduino\libraries\BME280\src/BME280I2C.h:53:16: note: virtual bool BME280I2C::ReadData(int32_t*)

   virtual bool ReadData(int32_t data[8]);

                ^

C:\Users\Steve\Documents\Arduino\libraries\BME280\src/BME280I2C.h:53:16: note:   candidate expects 1 argument, 5 provided

C:\Users\Steve\Documents\Arduino\libraries\BME280\examples\BME280_Modes\BME280_Modes.ino: In function 'void printBME280CalculatedData(Stream*)':

BME280_Modes:117: error: 'class BME280I2C' has no member named 'CalculateAltitude'

   float altitude = bme.CalculateAltitude(metric);

                        ^

exit status 1
no matching function for call to 'BME280I2C::ReadData(float&, float&, float&, uint8_t&, bool&)'

_____________________________________________________________________

finitespace, 2017-01-23:
Steve,

Looks like the Modes example never got updated. Give it a roll now. I corrected it.

Tyler

Bwanna, 2017-01-23:
Tyler,
Ok great. Thanks for letting me know.
Cheers,
Steve

closed 2017-01-26 by finitespace