finitespace/BME280 #17: Version 2.1.0

W4KRL, 2017-01-24:
Version 1.0.0 works fine but the IDE keeps nagging me to update to the latest 2.1.0 version. This is what happens with 2.1.0:

Arduino: 1.6.13 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
C:\Users\Karl\OneDrive\Arduino\libraries\BME280\src\BME280SpiSw.cpp:36:29: fatal error: Adafruit_BMP280.h: No such file or directory

 #include <Adafruit_BMP280.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R2 & mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

osos, 2017-01-25:
I have seen the same with platformio.

Workaround, delete the SPI related files and use the i2c

finitespace, 2017-01-26:
This issue has already been fixed in V2.1.01. #13 

Tyler

osos, 2017-01-27:
@finitespace, unfortunately it seems not to resolved with the version Platformio library collects:

`Compiling .pioenvs/d1_mini/lib/BME280_ID901/BME280Spi.o`
`Compiling .pioenvs/d1_mini/lib/BME280_ID901/BME280SpiSw.o`
`/home/johannes/.platformio/lib/BME280_ID901/src/BME280Spi.cpp:34:17: fatal error: SPI.h: No such file or directory`
`#include <SPI.h>`
`^`
`compilation terminated.`

I do not get why SPI is compiled when I only use i2c!

MacNetron, 2017-02-05:
I've got the same issue on v2.1.0, linux mint 18, Arduino 1.8.0:

`Arduino: 1.8.0 (Linux), Board: "Arduino Nano, ATmega328"

/home/mjfdreus/Arduino/libraries/BME280/src/BME280SpiSw.cpp:36:29: fatal error: Adafruit_BMP280.h: No such file or directory
 #include <Adafruit_BMP280.h>
                             ^
compilation terminated.
exit status 1
Error compiling for board Arduino Nano.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

The Arduino IDE does not have this library with a version higher than version 2.1.0  in the Manage Libraries Window.

BTW, as indicated above, v1.0.0 does work correctly on I2C.

coelner, 2017-02-16:
@MacNetron You need to download the latest version from here: https://github.com/finitespace/BME280/releases/tag/Release_Version_2.1.01
Arduino keeps moaning, that the version is unknown. To solve that issue, you need to change the version string in the library.properties back to 2.1.0

finitespace, 2017-02-17:
I just added a tag for Release 2.1.2. It might take a little for it to be incorporated into the Arduino libraries. But let me know if you can see it after about 24 hours.

Tyler

coelner, 2017-03-03:
Works for me.

closed 2017-03-06 by finitespace

MCeddy, 2019-04-24:
I have the same error like @psi-4ward 
After adding #include <SPI.h> it works on ESP32 :)

psi-4ward, 2019-12-15:
Issue still exists :\

teemuniiranen, 2019-12-30:
Yep, still existing with PlatformIO and Wemos D1 mini.

atar-axis, 2020-10-13:
Still an issue, unfortunately.
At least on the latest PlatformIO and 3.0.0

MichaelRChristensen, 2021-02-07:
> Still an issue, unfortunately.
> At least on the latest PlatformIO and 3.0.0

Just add.
#include <SPI.h>
in your main.cpp

Works on ESP8266 and Wemos D1 mini