finitespace/BME280 #24: 'NAN' error running in Energia

denzelwamburu, 2017-04-13:
Am trying to run the program in energia and am getting the following error. 
`BME_280_I2C_Test:78: error: 'NAN' was not declared in this scope`
`exit status 1`
`'NAN' was not declared in this scope.`

denzelwamburu, 2017-04-13:
am using the i2c code example

theopensourcerer, 2017-04-13:
I've not used Energia but I do use this in PlatformIO and it works fine. 

Have you tried it in an another IDE? 

What platform (MCU) are you building for?

finitespace, 2017-04-14:
denzelwamburu,

NaN (not a number) is a C++ 11 construct.  Do you have the capability in Energia to specify using the c++ 11 compiler? If not you may need to define NAN in the BME280 file as well as the function isnan();

Tyler

finitespace, 2017-04-14:
You can put something along the lines of:

#define NAN        1e+30  /* _HUGE_ENUF*/
#define isnan(n) n == float(NaN)

in the top of your BME280.cpp file.

Tyler

denzelwamburu, 2017-04-17:
Hi @finitespace , let me look at it tomorrow and give you the feedback. Thanks.

denzelwamburu, 2017-04-17:
@theopensourcerer MSP430

denzelwamburu, 2017-04-18:
@finitespace  Where exactly should this be placed?

finitespace, 2017-04-18:
At the top of the BME280.cpp file should work.

Tyler

Sent from iPhone

closed 2017-09-16 by finitespace