finitespace/BME280 #25: Absolute humidity measurement

milan-de, 2017-05-07:
Hi all,

as I am new to arduino and BME280 world I am not sure if my question I should ask under "Issues" or somwhere else, but in case I am on right place my question would be how to retrieve absolute humidity using this BME280 ? In library "finitespace/BME280" is mention only relative humidity, right ?
Thanks!

cheers,
Milan

coelner, 2017-05-26:
IMHO that is not a functionality of this sensor so it should not be part of the library either. There are thoughts about another library, maybe this code snippets helps you
`#include <math.h>

double calculationAbsHum(float t, float h)
{
  double temp;
  temp = pow(2.718281828, (17.67 * t) / (t + 243.5));
  return (6.112 * temp * h * 2.1674) / (273.15 + t);
}
String(calculationAbsHum(temperature, humidity)) + " g/m3");`

milan-de, 2017-07-12:
Thanks, is there any scientific reference on the Internet where this approximation is listed ?

coelner, 2017-07-13:
I'm not sure, but I guess here:
https://carnotcycle.wordpress.com/2012/08/04/how-to-convert-relative-humidity-to-absolute-humidity/
Otherwise:
http://www.vaisala.com/Vaisala%20Documents/Application%20notes/Humidity_Conversion_Formulas_B210973EN-F.pdf

closed 2017-07-15 by milan-de