finitespace/BME280 #26: Multiply sensors

milan-de, 2017-07-12:
Hi together,

is there any possibility to have 2 or more sensor in parallel connected, so to be able to masure on two different point humidity at the same time ? If yes, how to distinguish between sensors ? It seems that all are having address 0x76 per default.

reards,
Milan

finitespace, 2017-07-12:
Milan-de,

Yes, from the datasheet, "The last bit is changeable by
SDO value and can be changed during operation. Connecting SDO to GND results in slave
address 1110110 (0x76); connection it to VDDIO results in slave address 1110111 (0x77)..."

See this thread: 
https://forum.arduino.cc/index.php?topic=404946.0

Cheers!
Tyler

milan-de, 2017-07-13:
Thanks Tyler,

1. as I am using I2C interface I do not have access directly to SDO pin (like SPI interface) so I have to send some special message through SDA/SDI line right ?  
2. If so (1.) does your library support setting SDO values to Vddio/GND ? 
3. If (1.) and (2.) possible at all then I would have maximu of two sensors over one I2C bus right ?

regards,
Milan

finitespace, 2017-07-13:
Milan,

Yes, you would have a maximum of two devices. What breakout board are you using?

Tyler

milan-de, 2017-07-15:
Hi Tyler,

thanks for directing me, I figure out where is SDO line, but what I did not is by setting SDO pin I am changing master/slave relationship. So then one have situation of having one sensor master and one slave and arduino also as master (arduino is always master right?). So do we have conflict here ? Clock will be produced by both masters right ? 

regards,
Milan

closed 2017-07-15 by milan-de