finitespace/BME280 #27: consider changing Initialize() to public for bme280 forced mode

pabloandresm, 2017-08-25:
The forced mode of the bme280 allows a one shot read of the bme280 values (temp, humidity, etc).

The forced mode is selected in the constructor. The constructor calls Initialize() and sets forced mode (if the user set it to 1).

If that is the case, then the first read will give the values and put the bme280 in sleep mode (that is the way forced mode works).

The problem is that forced mode needs to be setted again to be able to be used again. It is resetted every read. The only way to do this is to call "WriteRegister(CONFIG_ADDR, config);" again.

Today it is called only at the constructor, which calls Initialize();

That is why I suggest to move in both I2C and Spi .h header files the Initialize() to "public:", so after a call to read we can call Initialize() again to set again forced mode and continue with another read and so forth.

That is the easy way. It works. I tested it.

Another more complicated way would be to handle it internally, with a flag somewhere, for every read, if the bme280 constructor setted forced mode, set it again before each read.

mclueppers, 2017-08-31:
It looks like you did some patches. Why not forking the project and creating PR later on? Way easier than just suggesting someone else to program it for you ;)

finitespace, 2017-08-31:
The constructor does not call initialize, begin does. For now, you can call begin to re-initialize the sensor.  I will think about if another function is required to provide different word choice when re-initializing. 

Tyler

closed 2017-09-20 by finitespace

pabloandresm, 2017-09-20:
I am currently developing the fix for this as you requested.
I don't think it is a good idea to close the bug just like that. It is a real bug. The specification says that if selected "forced mode", the after the temperature reading the forced mode is unsetted and you need to set it again. In this implementation the forced mode is setted only the begin() so it cannot be setted again.
Worse case you should leave it open, as it is a real bug.
I'll tell you when I finish the fix.
Pablo

finitespace, 2017-09-20:
Pablo,

I don't believe this is an issue. First off, the config reg does not contain the mode settings. These are held in the measure ctrl reg which can be reset at anytime by calling setMode. Second, I believe if no mode is specified it defaults to forced. Third, looking at the diagrams it says it returns to sleep after executing a forced read, but as previously stated I believe it defaults to forced on the next read. Finally, I don't believe the registers are modified, so any config settings will be preserved. All this in mind, while the datasheet shows that it returns to sleep mode, and it might be interpreted that the config needs to be updated every read,  I do no see any evidence that this is necessary in practice.

Tyler

finitespace, 2017-10-09:
Pablo, 

I did a massive refactor of the code this weekend and I actually found that is not an issue because it is already being done.  The config is pushed to the bme before every read if in forced mode.

Tyler

pabloandresm, 2017-10-30:
Thank you very much.