finitespace/BME280 #31: No Humidity Data?

bradyjoh, 2017-09-16:
I have a BME that seems to report temperature and atmospheric data without issue, but humidity data does not get reported. Any thoughts? I'm using the BME_280_I2C_Test code. 

Here's the Serial output: 

Temp: 27.22⸮C		Humidity: 0.00% RH		Pressure: 29.86 atm		Altitude: 18.16m		Dew point: nan⸮C

bradyjoh, 2017-09-16:
It looks like in the BME280.h file there is no return set for `float hum();` so could this be an issue, where the BME280 library is actually not reading the data?

idesignstuff, 2017-09-16:
You probably have a BMP280.
They don't do humidity.

Christian

bradyjoh, 2017-09-16:
I definitely have a BME280.

https://github.com/finitespace/BME280/blob/master/src/BME280.h#L90 is the line I'm talking about

finitespace, 2017-09-16:
You can read the chip id: chipID().

```
 static const uint8_t BME_ID          = 0x60;
 static const uint8_t BMP_ID          = 0x58;
```

Tyler

closed 2017-10-08 by finitespace