finitespace/BME280 #35: Error compiling 'BME_280_BRZO_I2C_Test.ino'

aderusha, 2017-10-09:
**Process:**

- Unzip arduino-1.8.4 to C:\
- Create `C:\arduino-1.8.4\portable`
- Add ESP8266 board manager URL, install ESP8266 board support, and select `NodeMCU 1.0 (ESP-12E Module)` from Board Manager
- Install library `Brzo I2C` Version 1.31 from Library Manager
- Install library `BME280` Version 2.2.0 from Library Manager
- Open example `BME280_BRZO_I2C_Test`
- Run "Verify" and the following error occurs:
```
Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\<user>\AppData\Local\Temp\arduino_cache_594038\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_8ee5fab911ac50cf47efcbc596507c23.a
sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `loop':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:84: undefined reference to `BME280I2C_BRZO::BME280I2C_BRZO(BME280I2C_BRZO::Settings const&)'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `__static_initialization_and_destruction_0':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:87: undefined reference to `BME280I2C_BRZO::BME280I2C_BRZO(BME280I2C_BRZO::Settings const&)'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `printBME280Data(Stream*)':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:88: undefined reference to `BME280I2C_BRZO::begin()'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `setup':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:51: undefined reference to `BME280I2C_BRZO::begin()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

finitespace, 2017-10-09:
Yes, I am aware of this bug. I am not sure why the compiler is not seeing the Brzo include define. Comment out/remove the #ifdef ifdef _BRZO_I2C_h at the top of the BME280I2C_BRZO.cpp and the #endif at the end. 

Tyler

aderusha, 2017-10-09:
In file `BME280I2C_BRZO.cpp` I've removed the `#ifdef _BRZO_I2C_h` at line 36 and the matching `#endif` at line 88.  No other changes from the test procedure outlined above.  Compiling the example file `BME_280_BRZO_I2C_Test.ino` now returns the following errors:
```
Archiving built core (caching) in: C:\Users\<user>\AppData\Local\Temp\arduino_cache_557126\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_8ee5fab911ac50cf47efcbc596507c23.a
sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `loop':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:84: undefined reference to `BME280I2C_BRZO::BME280I2C_BRZO(BME280I2C_BRZO::Settings const&)'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `__static_initialization_and_destruction_0':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:87: undefined reference to `BME280I2C_BRZO::BME280I2C_BRZO(BME280I2C_BRZO::Settings const&)'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `printBME280Data(Stream*)':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:88: undefined reference to `BME280I2C_BRZO::begin()'

sketch\BME_280_BRZO_I2C_Test.ino.cpp.o: In function `setup':

C:\arduino-1.8.4\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test/BME_280_BRZO_I2C_Test.ino:51: undefined reference to `BME280I2C_BRZO::begin()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

finitespace, 2017-10-10:
Pull down the latest version and try again. I wasn't getting the same compile errors as you, but I did fix some things. 

I was getting a bug with : 
'#define SDA_PIN = D2
#define SCL_PIN = D1'
it wasn't liking D1 & D2. I don't have time to debug it now, but I will look at it tonight. You might be able to get it working now.

aderusha, 2017-10-10:
Still no go.  Test procedure is now:
**Process:**

- Unzip arduino-1.8.4 to C:\arduino
- Create `C:\arduino\portable`
- Add ESP8266 board manager URL, install ESP8266 board support, and select `NodeMCU 1.0 (ESP-12E Module)` from Board Manager
- Install library `Brzo I2C` Version 1.31 from Library Manager
- Install library `BME280` from ZIP file downloaded ~ 11:00am GMT 2017-10-10
- Open example `BME280_BRZO_I2C_Test`
- Run "Verify" and the following error occurs:
```
Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\arduino\portable\sketchbook\libraries\BME280\examples\BME_280_BRZO_I2C_Test\BME_280_BRZO_I2C_Test.ino:32:0:

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:41:1: error: expected class-name before '{' token

 {

 ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:45:29: error: 'BME280I2C' has not been declared

    struct Settings : public BME280I2C::Settings

                             ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:45:40: error: expected '{' before 'Settings'

    struct Settings : public BME280I2C::Settings

                                        ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:66:13: error: 'Settings' does not name a type

       const Settings& settings = Settings());

             ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:66:43: error: ISO C++ forbids declaration of 'settings' with no type [-fpermissive]

       const Settings& settings = Settings());

                                           ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:60:4: error: invalid use of non-static data member 'BME280I2C_BRZO::Settings'

    };

    ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:66:34: error: from this location

       const Settings& settings = Settings());

                                  ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:48:7: error: 'OSR' was not declared in this scope

       OSR _tosr       = OSR_X1,

       ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:49:7: error: 'OSR' was not declared in this scope

       OSR _hosr       = OSR_X1,

       ^

C:\arduino\portable\sketchbook\libraries\BME280\src/BME280I2C_BRZO.h:50:7: error: 'OSR' was not declared in this scope
[... 117 lines cut ...]
    bme.read(pres, temp, hum, tempUnit, presUnit);

        ^

BME_280_BRZO_I2C_Test:80: error: 'tempUnit' was not declared in this scope

    bme.read(pres, temp, hum, tempUnit, presUnit);

                              ^

BME_280_BRZO_I2C_Test:80: error: 'presUnit' was not declared in this scope

    bme.read(pres, temp, hum, tempUnit, presUnit);

                                        ^

BME_280_BRZO_I2C_Test:84: error: 'BME280' was not declared in this scope

    client->print("Â°"+ String(tempUnit == BME280::TempUnit_Celcius ? 'C' :'F'));

                                           ^

exit status 1
call to 'BME280I2C_BRZO::BME280I2C_BRZO(const int&)' uses the default argument for parameter 1, which is not yet defined
```

finitespace, 2017-10-10:
Alright, I had committed the wrong file last night. 

With a fresh checkout as of 10/10/2017 10:26am MST:

-Modify line 35 BME280I2C_BRZO.cpp add: `#define USING_BRZO 1`
-Open example `BME280_BRZO_I2C_Test`
-Run "Verify".

I am getting a successful compile.

aderusha, 2017-10-10:
Confirmed working!  The example file now compiles with the instructions you've provided and I am now able to successfully read data from a test BME280 device connected to D1/D2 on a Wemos D1 mini. Thank you so much for your work on this and the crazy-fast turnaround on squashing bugs, I really appreciate it!

closed 2017-10-11 by finitespace

finitespace, 2017-10-11:
Outlined in #36.