finitespace/BME280 #40: rename sealevelAltitude

coelner, 2017-10-28:
rename sealevelAltitude to seaLevelPressure. It is a little bit misleading.

finitespace, 2017-10-29:
It returns altitude so I am not sure why you are suggesting the change in name.

coelner, 2017-10-29:
Maybe I understand that completely wrong, but the Altitude at SeaLevel is 0. 
But we need a function to get a SeaLevel correlated Pressure. So far I read the code, you calculate this. 
( https://de.wikipedia.org/wiki/Barometrische_H%C3%B6henformel#Internationale_H.C3.B6henformel )
(You need a SeaLevelAltitude while converting between different geodetic reference systems)

jirkaptr, 2017-10-29:
Hi, Tyler, it seems you have caught yourself in the trap of misleading name. The function sealevelAlitude really calculates the sea-level pressure which we know from meteorological reports. 
To the formula itself:
- I recommend to check the formula because it is not exactly the same as in the common sources (Wikipedia, NOAA). The speech is about presence of (0.0065 * A) in the denominator of the calculation formula.
- The formula does not respect air humidity.

I recommend the comparison with  http://drkfs.net/correctiontosealevel.htm.

finitespace, 2017-11-04:
Alright all, 

Good data. I should have required more documentation on the pull request. Let me try to clean this up.

Tyler

jirkaptr, 2017-11-05:
> Odesláno: sobota 4. listopadu 2017 10:27
> Komu: finitespace/BME280

Hello,

A lot of the time spent to compare calculation methods with official data of weather stations. 

Generally it is difficult to find complete set of data – usually the Pstation is omitted. I found complete data from Austria and Czech Republic only. 

https://www.zamg.ac.at/cms/de/wetter/wetterwerte-analysen 

http://portal.chmi.cz/aktualni-situace/aktualni-stav-pocasi/ceska-republika/stanice/profesionalni-stanice/mapy/teplota-a-vlhkost/stanice 

Regarding differences A vs. CZ: Each country uses a bit different algorithm.

 

The table is usable to check different calculation formulas and compare with official data. Secondary it is usable to calibrate pressure sensors. 

 

Note: Excel document is created in European language. If you need decimal “.”  instead of “,” in numbers then  you have to change the character…

 

Regards

 

Jiří Patera  (alias jirkaptr)

rpaskowitz, 2017-11-22:
I believe some renaming is in order. The readme claims that altitude is returned, but the [source](https://github.com/finitespace/BME280/blob/c50fa1510afb39affc4d1820e02d110aa8a5ff31/src/EnvironmentCalculations.h#L46) paints a more confusing picture:
```
  /// Convert current pressure to sea-level pressure, returns
  /// Altitude (in meters), temperature in Celsius
  /// return the equivalent pressure at sea level.
```

It says that it is converting pressures, then says it returns altitude in meters (and temperature), and then says it pressure at sea level. A simple test shows that it returns values that as much more in the range of pressure in `Pa` than altitude in `m`. It seems much more correct to say it *accepts* pressure (in Pa), Altitude (in meters) and Temperature (in Celsius), and *returns* pressure (in Pa). The `returns` on the first line of the comment seems erroneous and if left out, the comment as written makes far more sense (name of the method notwithstanding)

For example, a current reading from my BMP280 has a pressure of 101425 Pa, and the sealevel calculation returns 102105 Pa. Note that this is using a known-good altitude, and not the altitude as derived from the `Altitude()` method, which returns a significantly different value, but is computed with a default sealevel altitude of 101325 Pa as defined in the [header](https://github.com/finitespace/BME280/blob/c50fa1510afb39affc4d1820e02d110aa8a5ff31/src/EnvironmentCalculations.h#L43)

Given that this method seems to return a value in Pa, and that sealevel altitude should be pretty relatively constant, one might wonder what use this method has? For me, it seems to be for calibration of the `Altitude()`. When using the 101325 sealevel pressure, my altitude is about 70m from reality, so instead, using my known-good altitude, and obtaining the 102105 Pa pressure from the method, it seems I can use this value in place of the default sea-level pressure in `Altitude()` to get calibrated altitudes.

finitespace, 2017-11-22:
Here is the original pull request: #22

Looks like it was originally written to calculate equivalent sea level pressure in mb. When I refactored the code, my analysis was it was unit independent code, so, what unit you put in is what you will get out.

@guruathwal can you shed some light on the situation.

rpaskowitz, 2017-11-22:
I'd have to give it a more careful read, but that sounds correct. There winds up being a ratio of sealevel pressure to atmospheric pressure, and since the conversions between various units of pressure are linear, you should be able to pass in pressure in any unit, and get the adjusted pressure back out in the same unit.

It will just be pressure that it converted though, the temperature/altitude are fixed and used to determine the sealevel pressure.

finitespace, 2017-11-22:
Alright, a change is in order:
 - The function will be renamed `equivalentSeaLevelPressure`.
 - Documentation will be updated.
 - New/more functions can be added as mentioned above.

finitespace, 2017-11-22:
Completed: #49

closed 2017-11-22 by finitespace