finitespace/BME280 #45: Enhancement: Add HeatIndex

coelner, 2017-11-05:
we can calculate the heatindex from the rel. humidity and the temperature.
https://en.wikipedia.org/wiki/Heat_index

coelner, 2017-12-09:
a fast brain dump
```
int Heatindex
(
  float temperature,
  float humidity,
  bool metric
)
{
  float heatindex = NAN;
  float hi[2][9] = { {-8.784695,1.61139411,2.338549,-0.14611605,-1.2308094/100,-1.6424828/100,2.211732/1000,7.2546/10000,-3.582/1000000},
  {-42.379,2.04901523,10.1433127,-0.22475541,-6.83783/1000,-5.481717/100,1.22874/1000,8.5282/10000,-1.99/1000000} };

  //taken from https://de.wikipedia.org/wiki/Hitzeindex#Berechnung
  if (!isnan(humidity) && !isnan(temperature) && humidity>40 && (metric ? temperature>26.7 : temperature>80)) {
    heatindex = hi[metric ? 1 : 2][1];
    heatindex =+ hi[metric ? 1 : 2][2] * temperature;
    heatindex =+ hi[metric ? 1 : 2][3] * humidity;
    heatindex =+ hi[metric ? 1 : 2][4] * temperature * humidity;
    heatindex =+ hi[metric ? 1 : 2][5] * temperature * temperature;
    heatindex =+ hi[metric ? 1 : 2][6] * humidity * humidity;
    heatindex =+ hi[metric ? 1 : 2][7] * temperature * temperature * humidity;
    heatindex =+ hi[metric ? 1 : 2][8] * temperature * humidity * humidity;
    heatindex =+ hi[metric ? 1 : 2][9] * temperature * temperature * humidity * humidity;
    return int(heatindex);
  }
  else {
    return NAN;
  }
}
```

finitespace, 2017-12-11:
This look cool! If you can submit a pull request with documentation, that would be wonderful!

coelner, 2017-12-11:
#57

closed 2017-12-20 by coelner