finitespace/BME280 #59: First read after reset always 28.82 °C 81732.34 hPa

AndreKR, 2017-12-19:
Sketch:

```c++
#include <Wire.h>
#include <BME280.h>
#include <BME280I2C.h>

BME280I2C bme;

void setup() { 
  Serial.begin(115200);

  pinMode(17, OUTPUT);
  digitalWrite(17, 1);
  delay(50);

  Serial.println(F("Starting I2C"));
  Wire.begin(2,4); // SDA = GPIO2, SCL = GPIO4

  while (!bme.begin()) {
    Serial.println(F("Could not find BMx280 sensor!"));
    delay(500);
  }

  switch(bme.chipModel())
  {
    case BME280::ChipModel_BME280:
      Serial.println(F("Found BME280 sensor, measuring temperature, humidity and pressure"));
    break;
    case BME280::ChipModel_BMP280:
      Serial.println("Found BMP280 sensor, measuring temperature and pressure only");
    break;
    default:
      Serial.println("ERROR: Found unknown sensor");
  }
}

void loop() {
  float temperature = 0;
  float humidity = 0;
  float pressure = 0;
  
  bme.read(pressure, temperature, humidity, BME280::TempUnit_Celsius, BME280::PresUnit_Pa);

  Serial.print(temperature);
  Serial.print(" °C ");
  Serial.print(pressure);
  Serial.print(" hPa");
  Serial.println("");
  
  delay(500);
}
```

Output:
```text
Starting I2C
Found BMP280 sensor, measuring temperature and pressure only
28.82 °C 81732.34 hPa
25.78 °C 100944.22 hPa
25.79 °C 100947.60 hPa
25.78 °C 100943.45 hPa
25.78 °C 100944.22 hPa
25.78 °C 100944.22 hPa
```

coelner, 2017-12-20:
IMHO I think it is a bug and you can solve it by make two measurements.
(and you get Pa not hPa)

@finitespace 

- In BME280I2C.h you use filter_16, in all other settings Filter_Off
-  do we now how long a measurement takes? I can't find a value in the datasheet
- Maybe a problem with data register shadowing?
- [ ] add in the oversampling enum a zero value for disabling this feature.

finitespace, 2017-12-20:
Yes, Coelner is correct, I recently changed the Filter to 16 to provide more stable measurements (I think there is a typo in his post: I think he meant change the filter not the oversampling). 

The filter works by aggregating the last X samples together. So a larger filter requires more time to fill the buffer, when you reset, I would guess these values are set to their default, so it would make sense for the first read to give bogus data. I can go look at the datasheet later to see if they have any insight on this.

You can do one of two things:
1. Ignore the first read until the buffer has been populated. 
2. Set the filter to off. This will provide the instantaneous value instead of the more reliable aggregated value.

coelner, 2017-12-20:
@finitespace the part behind your name was only my thought while searching for the issue. Not directly related to this issue.

finitespace, 2017-12-27:
Alright, I have added a routine to initialize the filter buffer. Could you pull down the filtering branch and check if it fixes the problem? I don't have hardware with me right now.

finitespace, 2018-03-14:
I have created a pull request for this issue, I will leave it open for a little to allow review before I merge the changes. #81

closed 2018-03-14 by finitespace