finitespace/BME280 #61: getStatus function

coelner, 2017-12-21:
we should add a getStatus function where we select which bit is returned from the register.

finitespace, 2017-12-21:
I don't understand what this means.

coelner, 2017-12-22:
there is a status register at 0xF3 where bit 3 switch to '1' when a measurement is active. bit 0 shows only the copy of NVM data. (I'm not sure what conversion in this context means)
bit 3 is more important to know about, because we could check if a measurement is running and wait until the bit flips to '0' and read then the measured values.

finitespace, 2017-12-27:
This seems like it should be a private function that we check before returning a measurement. I wonder if we are returning the data before the filter buffer can be populated in #59.

finitespace, 2017-12-27:
I have added this function in the status branch. I need to do some testing on it to make sure it makes a measurement correctly while checking the status bits (see ReadData() for changes to flow). This might also cause the changes to #59 to be unnecessary.

finitespace, 2021-02-26:
We need to update the branch and create a PR.