finitespace/BME280 #63: Efficiency of heatindex calculation method

jirkaptr, 2017-12-21:
Calculation of the heatindex I believe works correctly, but I guess the function is not efficiently written. I'd recommend 

1. If necessary, convert the temperature to a opposite units and stop thinking which of coefficients tables to use. It will reduce the code size and increase the speed.

2. Do not use the stack or other RAM memory for the coefficients. Write coefficients directly into formulas. It will save up to (2tables*9coeffients*sizeof (float))= 72B of precious RAM. 

3. Perhaps it is possible to find a calculation method that will not be limited to 26.7 °C.

I apologize for the delayed reaction - only a very short time has elapsed since pull request to merge...

finitespace, 2017-12-21:
@coelner

coelner, 2017-12-22:
Yes, I take care of this.
@jirkaptr 
How did you do the measurement of speed and code size?
1. yes, it seems reasonable.
2. yes, but not by writing it directly to the formula. the values should be `const` including the use of PROGMEM. Why should it save RAM if we write it directly to the formula? 
3. No, because below this temperature the heatindex is not relevant. use windchill instead. I will add a textual reminder and return the original temperature.
4. Yes. I fixed that. Thanks.

jirkaptr, 2017-12-22:
@coelner:
**To point 2.**	
My idea regarding code efficiency improvement is this (substitution of lines 75 – 83):
    _heatindex = -8.784695;                        //coeff. 0
    heatindex += 1.61139411 * temperature;               //1
    heatindex +=  2.338549 * humidity;                   //2
    heatindex += - 0.14611605 * temperature * humidity;  //3
    float TempQ = temperature * temperature;    
    heatindex += (-1.2308094 / 100) * TempQ;             //4
    float HumQ = humidity * humidity;
    heatindex += (-1.6424828 / 100) * HumQ;              //5
    heatindex += (2.211732 / 1000) * TempQ * humidity;   //6
    heatindex += (7.2546 / 10000) * temperature * HumQ;  //7
    heatindex += (-3.582 / 1000000) * TempQ * HumQ;      //8_

The gist is in fact that constants written directly into code (also defined by #define) are (usually) treated as part of code, i.e. as constants taken directly into ALU from program code, and they do not occupy RAM. Another advantage is that there are not calculated and used pointers to array. This is most effective solution (but not ideally aesthetic).

**To point 3**.	
Most of resources provides information about limit 26.7/80. However I have found NOAA document http://www.wpc.ncep.noaa.gov/html/heatindex_equation.shtml . This document gives a good impression.

coelner, 2017-12-23:
@jirkaptr 
Point 2: Understood, but I think it is up to @finitespace to decide which way he wants
Point 3: https://github.com/adafruit/DHT-sensor-library/blob/09344416d2f3e8da0526494eb008c98d95fc0d79/DHT.cpp

finitespace, 2017-12-24:
Yeah, I would like to remove all these magic numbers from EnvironmentCalculations.cpp. I think the only way to do that without increasing memory usage is to to throw a bunch of defines at the top. That was an interesting solution to use PROGMEM, I am not sure of it's downfalls as I have not used it before.

coelner, 2017-12-24:
pull request contains NWS algorithm and coeffs as `#define`

finitespace, 2017-12-26:
@coelner @jirkaptr  I have merged @coelner into the heatIndex branch and made modifications to minimize the logic. Please check the logic again to ensure I didn't make any mistakes.

finitespace, 2018-03-14:
#65

closed 2018-03-14 by finitespace