finitespace/BME280 #83: Use of the functions

DavidRTucker, 2018-03-18:
Can you please explain how I can use setSettings to change the mode of the BME280. I guess was
bme.setSettings("Mode_Sleep") but the IDE knows that this is not valid. How do you do it?

finitespace, 2018-03-18:
BME280I2C_Modes.ino shows how to use the settings object, which is what is passed into the setSettings function. The README also has documentation on the Settings class.

DavidRTucker, 2018-03-19:
hank you for you very quick response!

Do you mean that I use the following two lines to change the OSR for pressure?

settings.presOSR = BME280::OSR_X16;
bme.setSettings(settings);

and  does the following code need to be placed above setup if I am changing the default settings?

BME280I2C::Settings settings(
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False,
   BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);

BME280I2C bme(settings);

Thank you for your help.  I can see that forced mode is very low power so I won't send it to sleep but I do want to oversample the pressure.

finitespace, 2018-03-24:
Yes, that is how you would set the settings. Forced mode goes back to sleep mode after a measurement and the default is sleep mode for the sensor, so putting it in sleep mode while using forced mode wouldn't change anything.