finitespace/BME280 #84: Error: no matching function for call to 'BME280::BME280()'

rbzg, 2018-03-29:
I'm trying to use the [Wemos-D1-Mini-BME280-Weather-Station](https://github.com/TheAustrian/Wemos-D1-Mini-BME280-Weather-Station) project . When compiling in Arduino IDE 1.8.5 I get the following errors:

Arduino:1.8.5 (Windows 10), Płytka:"WeMos D1 mini Pro, 80 MHz, 16M (15M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

In file included from C:\Users\rober\Documents\Arduino\meteo\meteo.ino:14:0:

settings.h:21: error: no matching function for call to 'BME280::BME280()'

 BME280 mySensor; // This can be tweaked, refer to https://github.com/finitespace/BME280 for details

sketch\settings.h:21:8: note: candidates are:

In file included from C:\Users\rober\Documents\Arduino\meteo\meteo.ino:7:0:

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:157:4: note: BME280::BME280(const BME280::Settings&)

    BME280(

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:157:4: note:   candidate expects 1 argument, 0 provided

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:43:7: note: constexpr BME280::BME280(const BME280&)

 class BME280

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:43:7: note:   candidate expects 1 argument, 0 provided

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:43:7: note: constexpr BME280::BME280(BME280&&)

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:43:7: note:   candidate expects 1 argument, 0 provided

In file included from C:\Users\rober\Documents\Arduino\meteo\meteo.ino:14:0:

settings.h:21: error: cannot declare variable 'mySensor' to be of abstract type 'BME280'

 BME280 mySensor; // This can be tweaked, refer to https://github.com/finitespace/BME280 for details

In file included from C:\Users\rober\Documents\Arduino\meteo\meteo.ino:7:0:

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:43:7: note:   because the following virtual functions are pure within 'BME280':

 class BME280

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:274:17: note: 	virtual bool BME280::WriteRegister(uint8_t, uint8_t)

    virtual bool WriteRegister(

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:280:17: note: 	virtual bool BME280::ReadRegister(uint8_t, uint8_t*, uint8_t)

    virtual bool ReadRegister(

C:\Users\rober\Documents\Arduino\meteo\meteo.ino: In function 'void loop()':

meteo:74: error: no matching function for call to 'BME280::ReadData(float&, float&, float&, bool, int)'

   mySensor.ReadData(pressureBuffer[bufferposition], tempBuffer[bufferposition], humidityBuffer[bufferposition], true, 0);   

C:\Users\rober\Documents\Arduino\meteo\meteo.ino:74:120: note: candidate is:

In file included from C:\Users\rober\Documents\Arduino\meteo\meteo.ino:7:0:

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:315:9: note: bool BME280::ReadData(int32_t*)

    bool ReadData(

C:\Users\rober\Documents\Arduino\libraries\BME280_finitespace\src/BME280.h:315:9: note:   candidate expects 1 argument, 5 provided

exit status 1
no matching function for call to 'BME280::BME280()'

I am very beginner, please help me solve the problem.

coelner, 2018-04-02:
1. You did not full fill the scheme for an issue.
2. It is not a problem in the library. The originator of your WeatherStation Sketch needs to fix that. ( I assume the project is abandoned)
3. Did you look at this? https://github.com/finitespace/BME280/blob/master/examples/BME_280_I2C_Test/BME_280_I2C_Test.ino
Especially line 23 and 28

finitespace, 2018-04-18:
You are using a newer version of the BME280 library and the prototype for the read function has changed. 

As @coelner  has pointed out, the example sketches show how to use the library.

The read function currently takes enumerations to the units:
```
   BME280::TempUnit tempUnit(BME280::TempUnit_Celsius);
   BME280::PresUnit presUnit(BME280::PresUnit_Pa);

   bme.read(pres, temp, hum, tempUnit, presUnit);
```

closed 2018-04-18 by finitespace

finitespace, 2018-04-18:
https://github.com/TheAustrian/Wemos-D1-Mini-BME280-Weather-Station/issues/4

marcdraco, 2020-12-13:
I got here with the same issue. I didn't use the .ino example because I'm using Platform IO with Visual Studio.

That specific example is right in the documentation under "Usage."

At least I know how to fix it which is something! :) but since I'm still very much on the upward part of the learning curve (C++) things like this throw me for a headscratcher.