finitespace/BME280 #85: compiler warning:

coelner, 2018-04-02:
### Expected behavior
return a bool from:
- bool BME280::InitializeFilter()
- bool BME280::WriteSettings()

### Actual behavior
```
/Arduino/libraries/BME280/src/BME280.cpp: In member function 'bool BME280::InitializeFilter()':
/Arduino/libraries/BME280/src/BME280.cpp:85:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/Arduino/libraries/BME280/src/BME280.cpp: In member function 'bool BME280::WriteSettings()':
/Arduino/libraries/BME280/src/BME280.cpp:122:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
```

### Steps to reporduce the behavior
compile the master branch

### suggested solution
bool BME280::InitializeFilter()
 1. read Data from Sensor to get individual boot up values
 2. Write Settings
 3. read Data from Sensor to get current values
 3. compare them

bool BME280::WriteSettings()
 0. Write Settings
 1. read Settings
 2. compare them
 3. return result

-- OR --
return the return value from used I2C functions

bertrik, 2018-08-07:
Please fix this, either doing complicated stuff like write-read-compare (as suggested here), or just return true (like in my pull request). The best compromise may be to simply return the i2c result, as a bool (success or not). I'm a bit surprised actually that something like this is classified by Arduino as a compile *warning*, I would consider this a hard *error* because currently the result is completely undefined.

finitespace, 2018-08-07:
Thank you for bring this to my attention @coelner. I have been extremely busy recently and haven't had time to look into this. My apologies. I will try to look at this tonight.

@bertrik the proper value needs to be returned or the function needs to be changed to void. I will take a look and see which is appropriate in this situation. 

Regards,
@finitespace

finitespace, 2021-02-24:
#117 is a duplicate, which I realized later, I have a new branch 117-non-void-return (or something like that)

@coelner @bertrik could one of you (or anyone else) test this branch, I don't have hardware right now, on biz trip. If it works as intended, then I will release 3.0.0. Finally! I need to pay attention to you guys more often. :)

finitespace, 2021-02-26:
#121

closed 2021-02-26 by finitespace