finitespace/BME280 #87: How does one set oversampling in an Arduino sketch?

DavidRTucker, 2018-05-10:
### Expected behavior

### Actual behavior

### Steps to reporduce the behavior

DavidRTucker, 2018-05-10:
I would just like to know how I can set the oversampling rate in my Arduino sketch

finitespace, 2018-05-10:
@DavidRTucker, we have already answered this in your last post. This is not an issue. Please stop opening issue requests. If you need more help, please continue to comment on the last issue. 

Part of being a good developer is learning how to read and use documentation. There is plenty of documentation and examples on how to set the settings. 

For your edification, to set the OSR:

```
BME280I2C::Settings settings(
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False,
   BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);

BME280I2C bme(settings);
```
For example if you wanted temperature to be x16 OSR you would write:
```
BME280I2C::Settings settings(
   BME280::OSR_X16,
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False,
   BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);

BME280I2C bme(settings);
```

Hopefully this helps.

DavidRTucker, 2018-05-26:
Thank you.  I have understood the settings function now.  In my sketch I define the units as BME280::PresUnit presUnit(BME280::PresUnit_Pa); but the values I get are hPa.  Have you noticed this?

Thanks again

finitespace, 2018-05-26:
I will take a look. There is both an hPa and a Pa unit.

closed 2021-02-26 by finitespace