finitespace/BME280 #88: **long** use instead of **float** use = possible?

esp13, 2018-05-26:
I have a question:
After reading the BME280 datasheet I resume this:
```
    Min     MAX     unité   précision   résolution
    -40     +85     °C      +/-0.5      0.01
    0       100     %Hr     +/-3%       0.008
    30000   110000  Pa      +/-100      0.18
```

So the pressure can move between this scale:
```
    Pression atmospherique
        30000.00    110000.00    => *100
        3000000     11000000=>  3x Bytes (ou 1 long)    => /100.0
        => xxxxxx.xx Pa
```
But this is a 8 significant digit.
A **float** variable can only goes for 7 significant digit. Would be better to use a **long** variable?

Whatever I read that it is preferable to use **long** variables instead of **float** ones (less use of processor cycles if no dedicated FTU)
And I am transferring it on NRF24L01 so in every cases I have to cut into **Bytes**.

So, Is there a way to get **long** instead of **float** when reading?

DavidRTucker, 2018-05-26:
The BME280 accuracy does not warrant 8 significant figures. A LONG requires the same 4 bytes as a FLOAT.

esp13, 2018-05-27:
Thanks for your answer,

Even with the .018Pa resolution? (I know the accuracy is only 100Pa)
[https://www.embeddedadventures.com/datasheets/BME280.pdf](https://www.embeddedadventures.com/datasheets/BME280.pdf)

What about the number of processor cycles to work with a float instead of working with a long?

the 11000000 (110000.00x100) maximum need only 3 Bytes to be send.

Without modifying anything it is not possible to get a double instead a float?

DavidRTucker, 2018-05-28:
I am plotting the difference between two BME280 modules that are at the same temperature. Overnight this difference gave rise to a metre of indicated height difference even with an averaging function over a period of 200 seconds. With barometric pressure readings speed is not required since it changes relatively slowly and hence noise can easily be eliminated. Perhaps my BME280s are from a bad batch and I should try a couple from Adafruit. I need more stability in the reading. I hope to find a better sensor than the BME280 that doesn't cost too much.

DavidRTucker, 2018-06-05:
The problem reading the BME280 with an Arduino is that a float only gives 6 significant figures. If you need more precision I you can minus 1000 hPa from the pressure readings. Then average over multiple readings. Averaging is essential to eliminate the noise which I find is rather higher than the stated 0.2 Pa claimed by Bosch.

finitespace, 2021-02-26:
I am fine moving it to a long. I would have to do some tests to see if there is any actual loss of precision.

finitespace, 2021-02-26:
Can you submit a PR for this?