finitespace/BME280 #93: BME280 or BMP280

Delivator, 2018-08-08:
### Expected behavior
See a humidity value

### Actual behavior
Sensor gets detected as BMP280 and it show 0.00% RH

I bought this sensor from ebay (https://www.ebay.de/itm/Blue-Digital-Barometric-Pressure-Sensor-Board-BME280-I2C-SPI-BMP280-3-3V-AIP/142636813700)  and thought that I bought a BME280 sensor, but now im not sure. How can I see if this is a BME280 or a BMP280? On the board and in the product title it says both.

gartfr, 2018-08-13:
You probalbly bough a BMP280 by error. 
The basic example program provided with this lib is able to differentiate the BMP from BME using the ChipModel_ function from the lib like this  : 

switch(bme.chipModel())
   {
      case BME280::ChipModel_BME280:
        Serial.println("Found BME280 sensor! Success.");
        break;
      case BME280::ChipModel_BMP280:
        Serial.println("Found BMP280 sensor! No Humidity available.");
        break;
      default:
        Serial.println("Found UNKNOWN sensor! Error!");

So, try to run this example (BME_280_I2C_Test or BME_280_SPI_Test depending your sketch) and check the first line of the serial output and you'll be fixed.

From my experience I ran BMP280 with this lib and I have this behaviour. RH=0.00%.

Delivator, 2018-08-13:
Yeah that's exactly what happened!
I contacted the seller and he confirmed that it was a BMP280.

closed 2018-08-13 by Delivator