finitespace/BME280 #94: SPI Mode Settings

gartfr, 2018-08-13:
Hello,

I'm playing with BME280 and this lib to get my project working. But, I'm running into difficulties.
Also, I apologize, because I'm quite a beginner at Arduino and BME.

My project consist of a BME280 connected to an Arduino Pro Mini using SPI interface.
The basic setting are perfecly working. Now Iwould like to play with modes Setting. So I inspired myself of the I2C example : 

BME280I2C::Settings settings(
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False,
   BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);
=> modified to this for SPI
#define DEVICE_PIN 9
BME280Spi::Settings settings(
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_True,
   DEVINCE_PIN  // SPI Device PIN
);

This is not compiling arguing that 1 parameter is expected but 8 are provided.
Maybe you can help me understanding the way to declare settings with SPI ? An example would be great !

So then I tried another way to play with settings and I use this method :

 // Change some settings before using.
settings.tempOSR = BME280::OSR_X4;
bme.setSettings(settings);
 
This is working OK. Even all OSR_X1 to X16 are working but not this even if it's documented:

 // Change some settings before using.
settings.tempOSR = BME280::OSR_Off;
bme.setSettings(settings);

Compiling says : OSR_Off is not part of BME280

Yep, I would like to disable some measurement of the BME280 in order to save power, and I won't use all three measurement available on the BME.

Thanks for your help ! :)

coelner, 2018-09-19:
Please look into the `src/BME280.h`, there is the missing part. I guess that you use the release 2.3.0 and not the current master.

gartfr, 2018-09-21:
Hello,
Thanks for your answer this saves my day !! :)

I finally got it working. You were right, I was running the last stable which is 2.3 and does not include all these options. I synced with the last master and changed a bit my code like this : 

#define DEVICE_PIN 9
BME280Spi::Settings settings(
   DEVICE_PIN,  //SPI Device Pin
   BME280::OSR_X1,
   BME280::OSR_X1,
   BME280::OSR_Off,
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False
);
BME280Spi bme(settings);  // Object Construction

Works as expected. If it can help someone else.
Best regards,

closed 2018-09-21 by gartfr