finitespace/BME280 #96: Lowest power mode possible

leestephens, 2018-08-24:
### Expected behavior
Looking in the [Datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-12.pdf), I think there is a mode (as described in 3.5.1) that does 1 sample per minute.

I assume this would be expressed in BME280.h setting enum StandbyTime as StandbyTime_60000ms

### Actual behavior

StandbyTime_1000ms is the largest value

### Request

Am I misunderstanding the way of using the code? Or can this be added? I actually only want to read the values every 10 minutes, so looking to get the lowest power possible. Maybe I might be better turning the device into Mode_Sleep and then switching to Mode_Forced when I want to take a reading?

BTW, thank you for an excellent library, very grateful for all the work involved.

coelner, 2018-12-19:
@leestephens gartfr answer is correct. the sensor can only sleep for 1000ms therefore use (as suggested) the forced mode. 
Could you close the issue?

leestephens, 2018-12-20:
Thanks once again.

closed 2018-12-20 by leestephens