finitespace/BME280 #98: Expose m_dig

Bolukan, 2018-10-28:
### Expected behavior
I would like to fingerprint each BME280 using the hash of the individual trimming parameters. I need a function that returns a copy of [uint8_t m_dig[32]](https://github.com/finitespace/BME280/blob/e1553a92191a2fb9f199789ba61b27e703233c13/src/BME280.h#L262).

### Actual behavior
Individual trimming parameter are not public.

finitespace, 2018-10-28:
Fell free to make a change and submit and pull request! I am not sure this is functionality the library should support, but, it never hurts to have it out there.

Thanks!

Bolukan, 2018-10-29:
Took right a shot: [Changes](https://github.com/Bolukan/BME280/commit/f317d2756f768e68a3eea046050734950b765f52)

The receiving array has to be supplied by reference. The function returns the reference to this array (so not void, see below an example).

Please comment, I am not proud on the long function name, but thought it should be descriptive.

```c
uint8_t parameters[32];
CRC32 crc;
uint32_t checksum = CRC32::calculate(bme.compensationParameters(parameters), 32);

Serial.printf("SensorID: 0x%08X\nHash(CRC32) of ", checksum);  
for (int i=0; i<32; i++) Serial.printf("%02X ", parameters[i]);  
Serial.print("\n");
```

Output:
```txt
SensorID: 0x27F4DEF5
Hash(CRC32) of CF 6C AF 65 32 00 54 91 5F D6 D0 0B 66 1D BC FF F9 FF 0C 30 20 D1 88 13 4B 47 01 00 1A 20 03 1E
```

Bolukan, 2018-11-23:
Little bump as your first reaction was so quick.

zoomx, 2018-11-26:
I believe that the simplest way is to add this in the library.
If I have understand first you have to call the ReadTrim private function and then you have all data in m_dig private array.
Then you can have a pointer to this array (so I believe it should be public) and calculate the CRC or calculate the CRC inside the library.

Internetwurm, 2019-12-22:
Hello Bolukan,

in the past i discovered this and it really works as suggested by Bosch.
https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Unique-IDs-in-Bosch-Sensors/td-p/6012

This is how you read the unique ID in any BME sensor.
As suggested it should work same way for BMP and also the 680 sensor.

Bolukan, 2019-12-22:
Fantastic. Let’s have this implemented !

Bolukan, 2019-12-27:
Implementation: https://github.com/Bolukan/BME280/

zoomx, 2019-12-28:
Example show an error here (IDE 1.8.10 and AVR core 1.8.2)
Serial.printf("Unique ID: 0x%08X\n", uniqueID);
'class HardwareSerial' has no member named 'printf';
print doesn't work too
no matching function for call to 'print(const char [19], uint32_t&)'

this compile but not tested 
`  uint32_t uniqueID = bme.UniqueId();
  char buffer[19];
  sprintf("Unique ID: 0x%08X\n", uniqueID);
  Serial.print(buffer);`

Bolukan, 2019-12-28:
You are right: printf is not available for all Arduino family members. I will use your comments to adapt the example. Thanks for testing!
For me, it worked: I got different numbers for different BME280's

zoomx, 2019-12-28:
I thought it was ESP8266 code, that's the reason I specified the AVR core.
Thank you for your unique ID add, it is very useful when you calibrate this sensor so you can easily associate calibration data with sensor as one can do with DS18B20.

Bolukan, 2020-01-05:
Example updated (Btw, 19 would be too small)

finitespace, 2021-02-26:
Has a pull request been created for this?