finitespace/BME280 #100: Sw I2C for BME280

cmoine, 2018-12-05:
Is there any particular reason I can't find a BME280 driver that use software i2c ?

You'll probably tell me: why don't you use SPI ? Because I have 3 sensors that only accepts i2c :( :(

Why software i2c ? Because my Arduino is already a I2C slave of a Raspberry :)

I already worked on adapting a BME280 library so use software I2C, but unfortunately, the version doesn't compile with newer version (that is the problem when we "fork" a project).

Would you eventually be interested in a Pull Request for this ?

cmoine, 2018-12-05:
Ups, I guess need to combine this library with Brzo ? :)

idesignstuff, 2018-12-05:
I've only ever used this lib with I2C, for years.
It's running with 3v3,Gnd,SCL,SDA wires on several Wemos D1 minis, flashed
with either platformio in Atom, or Arduino IDE.
I've had 2 BME's wired to the same SDA and SCL pins, with two different I2C
addresses, and it worked flawlessly.

Christian

cmoine, 2018-12-05:
Problem with Brzo:
WARNING: library Brzo_I2C claims to run on (esp8266) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
And it doesn't even build :-(

idesignstuff, 2018-12-05:
No need for BRZO. I2C is built into BME280 by FiniteSpace.

Christian

idesignstuff, 2018-12-05:
https://github.com/finitespace/BME280#settings
Christian

cmoine, 2018-12-05:
Thank you Christian, but I2C for Arduino Nano is bounded to A4 & A5.

How do I use D2 & D3 for instance ?

Only solution remaining I guess is using this:
https://github.com/felias-fogg/SoftI2CMaster

finitespace, 2018-12-05:
Yeah, Brzo needs to be discontinued. Unless you are running an esp, you shouldn’t need it.

finitespace, 2018-12-05:
Thank you everyone for responding to this. I know I have been amiss in my duties this year. Hopefully, I will have some time over Xmas break to contribute to changes.

I value the support group this library has gained.

Regards,
Finitespace or should I say, finitespacetime ;)

finitespace, 2018-12-05:
> We already support I2C. No need for a pull request.
>
> Do you already support software I2C !? Do you have an example ?

Implement your solution. I will take a look.

cmoine, 2018-12-06:
But wait a minute, I looked at Git  branches, and one is named softwarewire, which make use of SoftwareWire.h... A library for software I2C on any pin: this is exactly what I need ? Is this working ? Any plan to integrate it in the main branch ?

cmoine, 2018-12-07:
It works ! however, I encounter a GCC segmentation fault when I comment one of these lines
```
 bme.read(pres, temp, hum, tempUnit, presUnit);
  ...
  Wire.write("{\"temperature\":");
```
GCC seg fault with arduino are very anoying :( :( Any tip ?

cmoine, 2018-12-07:
Ok found the issue GCC:
http://forum.arduino.cc/index.php?topic=583509

cmoine, 2018-12-07:
Ok it seems to work, but still some problem to fix on my side, and it seems that I have burnt my Arduino somehow, so my tests will ahve to wait next Arduino order :) Is it possible to merge/publish this ?

closed 2021-02-26 by finitespace