finitespace/BME280 #102: Parsing H4/H5 humidity calibration data

AmedeeBulle, 2018-12-29:
According to the datasheet, dig_H4 and dig_H5 are signed integers on 12 bits

The following code handles these as pure 16bits signed, which will never be negative...

https://github.com/finitespace/BME280/blob/7a211f03aa3ac5567b14e2cbc12ac21e185b6e0b/src/BME280.cpp#L277-L278

The datasheet is ambiguous as it refers to these 12 bits values as `signed short`, but the [BoschSensortec reference code](https://github.com/BoschSensortec/BME280_driver/blob/a50e719dedec9c08efc3c8a9708881dfc4940848/bme280.c#L1227-L1233) definitely handles the values as 12 bits signed integer -- the `(int8_t)`cast on the msb together with a multiplication instead of a shift ensure the sign is propagated properly to the `int16_t`

Bolukan, 2018-12-29:
formula:
x = (x >> 11) == 0 ? x : -1 ^ 0xFFF | x;

finitespace, 2019-01-26:
This code was taken directly from the Bosch example code. 

Have you seen adverse affects due to it?

AmedeeBulle, 2019-01-26:
I am not sure which Bosch example code you are referring to, but in all the Bosch code I have seen, dig_H4 and dig_H5 are taken as __signed__ integers on 12 bits.

On the sensors I have, dig_H4 and dig_H5 are positive so it doesn't make a difference, but it would if they were negative.

Bolukan, 2019-01-26:
Don't they cast it to signed?
[dig_H4_msb = (int16_t)(int8_t)reg_data[3] * 16;](https://github.com/BoschSensortec/BME280_driver/blob/a50e719dedec9c08efc3c8a9708881dfc4940848/bme280.c#L1227)

AmedeeBulle, 2019-01-27:
Yes they do, and this is what I am proposing in #103

closed 2021-02-26 by finitespace