finitespace/BME280 #105: EnvironmentCalculations SeaLevelPressure

Gonzo4, 2019-05-26:
Hi,

just a question, from where did you get the formula to calculate the EquivalentSeaLevelPressure?
The number 5.257 in your formula is always 5.255 googeling around, like in Wikipedia or like here https://www.sparkfun.com/tutorials/253

jirkaptr, 2020-03-08:
The EquivalentSeaLevelPressure formula was taken from http://keisan.casio.com/exec/system/1224575267. 
Some sources indicate the constant 5,255, others 5,257. It all depends on what value is used for the vertical air temperature gradient (**a**), and how accurate is the calculation of the constant **M * g / R * a**., where 
**M** ... die mittlere moler Masse der Atmosphärengase (0,02896 kg/mol),
**g** ... die Schwerebeschleunigung (9,807 m/s^2),
**R** ... die universelle Gaskonstante ( 8,314 J K^(− 1) mol^(−1) )
The German Wikipedia states: "Für einen Temperaturgradienten von 0,65 K für 100 m nimmt der Exponent den Wert 5,255 an".

finitespace, 2021-02-26:
Was this resolved?

Gonzo4, 2021-02-26:
Yes, could be resolved.
@jirkaptr has explained it very well, I had thought it was just a typo 👍

closed 2021-02-26 by Gonzo4