finitespace/BME280 #106: error: 'I2CAddr_0x76' is not a member of 'BME280I2C'

MacSass, 2019-07-21:
Helllo,

been using the library in the past successfully, but can´t get my code to compile anymore.
Also the example .ino BME280_Modes.ino does not compile anymore more for - always throws an error: 
test:62:4: error: 'I2CAddr_0x76' is not a member of 'BME280I2C'

Unfortunately I´m not too savy to figure out why this is. It compiles fine when I remove the line
BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.

So I might be good with using the default address - but wondering why that does not compile anymore, in case I need a different address.

Thanks for your help - MacSass

RichieRogers, 2019-07-30:
I get that issue on a basic test I'm doing.
The instructions were a bit vague on how to configure for different addresses, but I found this:
BME280I2C::Settings settings(
   BME280::OSR_X1, // Temperature
   BME280::OSR_X1, // Humidity
   BME280::OSR_X1, // Pressure
   BME280::Mode_Forced,
   BME280::StandbyTime_1000ms,
   BME280::Filter_Off,
   BME280::SpiEnable_False,
   BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.
);

BME280I2C bme(settings);

However, setting the address generates this error on compile:
BME280Test:37:4: error: 'I2CAddr_0x76' is not a member of 'BME280I2C'

    BME280I2C::I2CAddr_0x76 // I2C address. I2C specific.

    ^

exit status 1
'I2CAddr_0x76' is not a member of 'BME280I2C'

Is there something I (or we) are doing wrong?

Thanks,
Richie

RichieRogers, 2019-07-30:
Right, found the answer.
Replace: BME280I2C::I2CAddr_0x76 with 0x76
or BME280I2C::I2CAddr_0x77 with 0x77

Richie

MacSass, 2019-07-30:
Hi @RichieRogers ,
so the code would like the following then?

BME280I2C::Settings settings(
BME280::OSR_X1, // Temperature
BME280::OSR_X1, // Humidity
BME280::OSR_X1, // Pressure
BME280::Mode_Forced,
BME280::StandbyTime_1000ms,
BME280::Filter_Off,
BME280::SpiEnable_False,
0x76 // I2C address. I2C specific.
);

I´ll give that a try ... 

thanks - hope the orignal issue get´s fixed.

Regards - MacSass

RichieRogers, 2019-07-30:
Hi,
Well, it worked for me on a test board I've just built :)

Thanks,
Richie

closed 2021-02-26 by finitespace