finitespace/BME280 #108: 2 BME280 sensors (I²C 0x76/0x77)

PeterPitzeier, 2019-09-15:
Is it possible to read 2 BME280 sensors (I²C 0x76/0x77) ?

Bolukan, 2019-10-13:
As the I2C address is part of the initial settings, you can initiate two instances with different addresses. So the library seens ready for your use case.

markrhickman, 2021-02-23:
I wanted to design a class that accepted an i2c address as an argument for the class constructor.  The class was already a derived class from a generic weatherstation class.  For this reason, and my relative inexperience with classes in general I am currently not making BME280I2C a base class to the new class.

I do, however create a default BME280I2C::Settings structure  (m_settings) and a BME280I2C bme( m_settings) instance for each of the new class instances.  This requires the constructor for the derived class to be able to update the private variable BME280I2C::m_bme_280_addr.  this is not possible in the current version of BME280I2C.  m_bme_280e_addr is only initialized by the constructor. 

Furthermore, in an attempt to troubleshoot, the BME280::getSettings() is ignorant of the bme280addr member added to the Settings structure.

How I found I could meet my goal required the following steps:
1. make BME280::WriteSettings() protected rather than private
2. define BME280I2C::getSettings() so that bme280Addr is visible.  The action is identical to BME280::getSettings() other than using the derived Settings structure rather than the base structure.
3. define BME280I2C:setSettings() which calls BME280::WriteSettings, necessitating WriteSettings() being protected, then setting m_bme_280e_addr to settings.bme280Addr.

I have just gotten this working so I do not know if I have caused any other problems.  I used this problem as a learning exercise because I was not familiar with structures and struct constructors and only vaguely familiar with classes and class constructors.  Of course, that means I am blissfully unaware of any coding sins I may have committed.

If anyone is interested I will create a fork and submit a pull request.

finitespace, 2021-02-23:
First. don't create a fork, create a branch and then submit a pull request.

It's hard to visualize what you are saying here, if you are able to create a pull request, I can better address the changes.

markrhickman, 2021-02-23:
I just cloned the repository and realized that I was working with an older version.  I am going to review the changes between the versions before I add my own.  (I now have the opportunity to understand the "virtual" keyword).  I will my branch if it still seems relevant.

Regards,
Mark

markrhickman, 2021-02-23:
The current version from the repository solved the problems I was solving.  I only needed to change my code for the new enumeration of the i2c addresses.  There is no need for me to submit my changes.  Thank you.

FYI my original version came from the arduino library and was version 2.3.0 rather than 3.0.0 of the master branch.
Have a good day and thank you for responding quickly.

Mark

closed 2021-02-26 by finitespace