finitespace/BME280 #113: Filter is only applied to temperature and pressure

FrederikWolter, 2020-05-16:
### Expected behavior
filter set with BME280 :: Filter_8  
So it should be filtering temp, hum and pres; right? 

### Actual behavior
Temp and pres seem pretty smooth:
[image]

[image]

But humidity isn´t:
[image]

### Steps to reporduce the behavior
I used the following code with library version 2.3.0 on an Wemos D1 mini:

> BME280I2C::Settings settings(
>   BME280::OSR_X1,                  
>   BME280::OSR_X1,                  
>   BME280::OSR_X1,                  
>   BME280::Mode_Forced,            
>   BME280::StandbyTime_1000ms,
>   BME280::Filter_8,               
>   BME280::SpiEnable_False
> );
> BME280I2C bme(settings);

> void setup()
> {
>   Serial.begin(9600);
>   while (!Serial) {}
> 
>   // Sensors  
>   Serial.print("Starting Sensor ...");
>   Wire.begin(4, 5);                  
>   while (!bme.begin()) {
>       Serial.println("waiting Sensor ...");
>     delay(50);
>   }
>     Serial.println("Sensors_done!");
> }

> void loop()
> {
>   float temp(NAN), hum(NAN), pres(NAN);
>   bme.read(pres, temp, hum, BME280::TempUnit_Celsius, BME280::PresUnit_hPa);
> 
>   Serial.println(temp);
>   Serial.println(hum);
>   Serial.println(pres);
>   delay(250000);
> }

finitespace, 2021-02-26:
Hello, if you would like to try and solve this and submit a pull request, that would be awesome!

finitespace, 2021-02-26:
Anyone else know why this might be? I have been away from the code for a while

david-lutz, 2021-03-20:
Section 3.4.4 of the Datasheet (https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf) would seem to imply that the BME280 doesn't do IIR filtering for the humidity sensor.

closed 2021-04-17 by finitespace