finitespace/BME280 #114: STM32F103C8 - bme.begin (); - program stops

EmilM256, 2020-05-28:
My program stops when I have a BME280 sensor connected to the board via I2C. Exactly on the command:
`bme.begin ();`
If the sensor is not connected to the board, the program works correctly, it does not stop. I get the message on the serial port:
`Could not find BME280I2C sensor!`
repeated every second. I used the example BME280_Modes.

This happens on the configuration:
```
MCU: STM32F103C8 blue pill
Arduino IDE version 1.8.12
Official Core 1.9.0
Library version 2.3.0
```

Interestingly, the sample sketch works correctly on the configuration:
```
MCU: STM32F103C8 blue pill
Arduino IDE version 1.8.12
Roger's core 2020.5.16
Library version 2.3.0
```
It also receives correct measurement results.

Unfortunately, I would like the library to work with the Official Core due to other peripherals and libraries.
Please help.

finitespace, 2021-02-26:
Did you try calling Wire.begin() before bme.begin()?

tkalfaoglu, 2022-10-04:
I have the same issue -- wire.begin() works fine but it hangs at bme.begin()

finitespace, 2023-03-02:
Its really hard for me to give any useful advice without know what hardware you are running on, examples of code, etc