finitespace/BME280 #115: BME280 v3.0 not work with u8g2 and display HX1230_SW_SPI

ruzanow, 2020-09-06:
Sensor BME280 (with default address) connected to arduino I2C.

First - library BME280 v2.1 work without problem.

BME280 v3.0 normally work with graphics library [u8g2](https://github.com/olikraus/u8g2) and OLED screens on I2C.
But. BME280 v3.0 not work with u8g2 and LCD screen HX1230 on software SPI.

### Expected behavior
With OLED: BME280.cpp 
 BME280::ReadChipID() 
   Serial.println(id[0]) = 96

### Actual behavior
with LCD: BME280.cpp

  BME280::ReadChipID() 
    Serial.println(id[0]) = 10

### Steps to reporduce the behavior

Here [sketch](https://github.com/ruzanow/simply-meteo/blob/master/sketch.ino) Uncomment U8G2_HX1230.

ruzanow, 2020-09-06:
Edit ReadChipID to always return true - no effect
Old function ReadTrim() - no effect ( ord == 0 )
Add Wire.begin() to old ReadTrim() - partially work

ruzanow, 2020-09-06:
Wire.begin(); in BME280I2C.cpp finally fixed this issue.

finitespace, 2021-02-26:
Yes, the user needs to call Wire.begin()

closed 2021-02-26 by finitespace