finitespace/BME280 #118: Propose License Change to LGPL

tinxx, 2020-12-30:
Hello,
First-off let me say thanks for the work you put into creating this library.

I was looking through the libraries I include in my current project and had to realize that this library would force me to release my code under GPL, too. I don't say that GPL is a bad licence, actually I believe the opposite.

Anyhow, I suggest a license change to LGPL to make it easier for people to integrate the library into projects licensed differently.

Cheers,
tinxx

finitespace, 2021-02-26:
Hmm, I would prefer it to go to MIT, but I would need by off from all contributors.

If you can take the time to get everyone on board, I am happy to change.

finitespace, 2021-02-26:
@coelner 
@jirkaptr 
@klagr 
@HappyWheels 
@chacal 
@guruathwal 
@alexshavlovsky 
@johanso-au 
@rpaskowitz 
@eykamp 
@jamesmyatt 
@vortigont 
@osos 
@per1234 
@maru-sama 
@AmedeeBulle 
@Amunak 

How would you feel about a license change?

per1234, 2021-02-26:
@finitespace you have my permission to do anything you want with my insignificant contribution.

HappyWheels, 2021-02-26:
@finitespace You have my permission to change the license as you see fit.

AmedeeBulle, 2021-02-26:
I support the change to a less restrictive license (LGPL or other).

While GPL is in general a good thing, delivering a **library** under GPL is quite restrictive as it is not compatible with most of the other open source licenses libraries you might want/need to include in your project.
(In fact I am not using this library for some of my projects just because of a license conflict with other libs...)

guruathwal, 2021-02-26:
@finitespace I do not have any objection to changing the licence to any other type.

Amunak, 2021-02-26:
I'm all for it, sure; feel free to license my one-word contribution under any license you wish, now and in the future. ;)

vortigont, 2021-02-26:
I'm OK! LGPL might be more suitable for a lib in some cases, so be it :)

maru-sama, 2021-02-26:
This is OK for me too.

eykamp, 2021-02-26:
I think the LGPL would be preferable to the GPL, but I think MIT or similar would be even better for a library.

But regardless, I consent to this change.

jirkaptr, 2021-02-26:
@finitespace,  I don't have a problem with that.

coelner, 2021-02-27:
So far (myself included)
- [x] coelner
- [x] jirkaptr
- [ ] klagr
- [x] HappyWheels
- [ ] chacal
- [x] guruathwal
- [ ] alexshavlovsky
- [ ] johanso-au
- [ ] rpaskowitz
- [x] eykamp
- [ ] jamesmyatt
- [x] vortigont
- [ ] osos
- [x] per1234
- [x] maru-sama
- [x] AmedeeBulle
- [x] Amunak

chacal, 2021-02-27:
Fine by me! 👍

alexshavlovsky, 2021-02-28:
I don't mind. Feel free to do whatever you need with the licensing issue

osos, 2021-03-01:
Fine with me.

rpaskowitz, 2021-03-07:
Either LGPL or MIT are good with me.

tinxx, 2021-03-29:
Hey folks,
Let me summarize the current state of the license change proposal:

I initially proposed LGPL as the new license because the original license was GPL.
@finitespace, however, is the main author and proposed MIT as the new license.
MIT would be even more permissive and would enable the library to be used more easily and extensively.

Now there are some unclear and missing statements from contributors that I would like to get cleared up.

## Unclear Contributors

I would like to ask the following contributors to clarify their decisions:

* @chacal,
* @coelner
* @jirkaptr
* @maru-sama
* @osos: 

Do you support either MIT, LGPL, both or basically any license?

@vortigont, you signalled you are OK with LGPL. Are you OK with MIT license, too?

## OK Contributors (LGPL + MIT)

The following contributors gave their OK for both LGPL and MIT licenses (or _any_ license):

* alexshavlovsky
* AmedeeBulle
* Amunak
* eykamp
* guruathwal
* HappyWheels
* per1234
* rpaskowitz

## Other Contributors

* @johanso-au has so far not been heard from.
* @yanbec has not been mentioned here so far. However I found his name on: `BME280.h` (165-168, 253-255, 315); `BME280.cpp` (154-161)

@johanso-au, @yanbec, please tell us if you are OK with a license change to MIT and/or LGPL license?

---

Sorry for the inconvenience!
--Tinxx

chacal, 2021-03-29:
All licenses are fine for me!

osos, 2021-03-29:
I am OK with any license.

vortigont, 2021-03-29:
I'm also OK with any license.
Cheers!

coelner, 2021-03-30:
LGPL or MIT are fine.

finitespace, 2021-04-17:
Alright, it looks like we have quorum! I am going to assume anyone who has not already spoke up has abandoned their account - so speak up now or forever hold your silence. 

For the remaining unspoken - I will take your silence as consent

Would someone volunteer to create a new merge request @coelner @tinxx? I would personally like to change it to MIT.

tinxx, 2021-04-23:
Hi @finitespace,

I have created a PR changing all license annotations to `Copyright (c) 2015-2021 Tyler Glenn, et al.`.
Where there were explicit mentions of other contributors (`Forked by Alex Shavlovsky`), I have also added the person to the Copyright notice.

Cheers!
Tinxx

tinxx, 2021-06-04:
Hey @finitespace, any news?

finitespace, 2021-07-26:
Thank you, @tinxx. That was an excellent pull request, are you looking for work? ;) A few minor comments, let me know your thoughts,

Kind regards, and sorry for the delay,
@finitespace

tinxx, 2021-08-18:
Hey @finitespace,

Thank you for the warm response.

I have pushed the amendments according to your comments. I hope this is to your satisfaction ;)
 
You can always contact me privately with a job offer ;)

Cheers!
@tinxx

tinxx, 2022-02-05:
Bump! Please re-visit the Merge Request! Thanks!