finitespace/BME280 #129: non-void-return issue in 3.0.0

dhzl84, 2021-08-24:
### Expected behavior
Version 3.0.0 should not throw return type errors fixed with #121.

### Actual behavior
Eventhough PlatformIO states that it downloaded version 3.0.0, the functions in the downloaded code still have the return type _bool_.

Builds fail locally as well as on [GH Actions](https://github.com/dhzl84/ESP8266-Smart-Home/runs/3410614436?check_suite_focus=true).

### Steps to reporduce the behavior
I use PlatformIO to get the library.

platformio.ini:
```
lib_deps = 
	finitespace/BME280 @ ^3.0.0
```

Library Installation by PlatformIO:
```
Library Manager: Installing finitespace/BME280 @ ^3.0.0
Downloading...
Unpacking...
Library Manager: BME280 @ 3.0.0 has been installed!
```
PlatformIO build console:
```
Dependency Graph
|-- <BME280> 3.0.0
```
```
.pio\libdeps\8266_thermostat_debug\BME280\src\BME280.cpp: In member function 'bool BME280::InitializeFilter()':
.pio\libdeps\8266_thermostat_debug\BME280\src\BME280.cpp:85:1: error: no return statement in function returning non-void [-Werror=return-type]
   85 | }
      | ^
.pio\libdeps\8266_thermostat_debug\BME280\src\BME280.cpp: In member function 'bool BME280::WriteSettings()':
.pio\libdeps\8266_thermostat_debug\BME280\src\BME280.cpp:122:1: error: no return statement in function returning non-void [-Werror=return-type]
  122 | }
      | ^
```

There seems to be an issue in the library registry.
Version 3.0.0 has two release dates, maybe same version published twice?
I browsed the history of my _platformio.ini_ and already used version 3.0.0 in January 2020.

[image]

and

[image]

Hope you find the time to have a look into that?

MacSass, 2021-10-18:
Manual download and install to platformio fixed the issue ... looks like the old version is still on the repository!

villafapd, 2021-12-05:
I had the same problem and I solved it by manually downloading the library.

lekimtek, 2021-12-13:
The same for me

lekimtek, 2022-01-04:
Update!
I installed manually  "src" dir from branch 3.0.1-dev and not work, same error like installed by platformio.
Aclaration:

This not work and may be this is installed by platformio instead 3.0.0
git clone -b 3.0.1-dev https://github.com/finitespace/BME280.git

This work fine if is installed manually not by platformio.
git clone https://github.com/finitespace/BME280.git

Where is the error?
May be platformio search latest  and install 3.0.1-dev?

studiofuga, 2022-07-29:
See https://github.com/finitespace/BME280/issues/130#issuecomment-1199143984 on issue #130 

Solution: Publish an updated version on platform.io.

Workaround: specify the git URL on platformio.ini:

```
lib_deps = 
            BME280=https://github.com/finitespace/BME280.git

```