finitespace/BME280 #130: ESP32 boot loop when calling bme.begin()

chikiball, 2022-05-05:
First off, did you read the manual? Did it answer your question? Yes, No
Second, did you look at the examples? Did they answer your question? Yes, No
No, alright, please describe the problem below.

I am using ESP32 Dev Kit. All wirings are confirmed.
When calling bme.begin(), ESP32 goes into panic mode and keep on rebooting.

### Expected behavior
Should not reboot

### Actual behavior
Goes into boot loop

### Steps to reporduce the behavior
call bme.begin() in the setup() function.

chikiball, 2022-05-05:
Solution:

BME280.cpp InitializeFilter() should be a void function instead of expecting to return bool.
BME280.cpp WriteSettings() should be a void function instead of expecting to return bool.

Downloaded the library from platformIO library, I guess it is not the latest update.

biterrorr, 2022-06-15:
The same problem also affects ESP8266 with board versions 3.0.x (with 2.7.4, bme.begin() does not crash).  Changing these functions to void type seems to fix the problem.

studiofuga, 2022-07-29:
The issue was fixed in branch `master` since Feb 11th, but an updated version wasn't published on platform.io, so adding finitespace/BME280 on `platformio.ini` results in downloading a buggy version. This explains why this only happens on platformio and not on arduino or by downloading from github.

```
$ pio pkg show finitespace/BME280
finitespace/BME280
Library • 3.0.0 • Public • Published on Wed Nov 22 07:39:41 2017

Provides a library for reading and interpreting Bosch BME280 environmental sensor data over I2C, SPI or Sw SPI. Reads temperature, humidity, and pressure. Includes environment calculations. Provides functions for english and metric. Also reads pressure in Pa, hPa, inHg, atm, bar, torr, N/m^2 and psi. ESP and BRZO I2C support.

---------------------  ------------------------------------------------------------
Registry               https://registry.platformio.org/libraries/finitespace/BME280
Homepage               https://www.github.com/finitespace/BME280
Repository             https://github.com/finitespace/BME280.git
License                GPL-3.0
Popularity             273
Stars                  172
Examples               6
Installed Size         109.75KB
Used By                1
Compatible Platforms   *
Compatible Frameworks  arduino
Keywords               sensors
---------------------  ------------------------------------------------------------

Version    Size     Published
---------  -------  -------------------
3.0.0      25.05KB  2017-11-22 07:39:41
2.2.1      24.67KB  2017-10-11 07:55:12
```

Solution: Publish an updated version on platform.io.

Workaround: specify the git URL on `platformio.ini`:

```
lib_deps = 
            BME280=https://github.com/finitespace/BME280.git
```